bigtreetech/BIGTREETECH-SKR-V1.3 #1: smoothieware does support Linux

gdachs, 2019-03-03:
Sorry, but this sentence in the instructions for the SKR 1.3 makes no sense:

> Note: When using smoothieware open-source firmware, since only Windows system
> drivers, the SKR board at this time can only be recognized by the Windows system
> port! ! !

If you didn't find Linux driver, then you searched for them at the wrong place. Linux has drivers for all known USB to seriell drivers already builtin. Currently I don't own a SKR 1.3 board, but I use the MKS Sbase with smoothieware with Linux without problems. The SKR and the Sbase are very similar, at least they use the same CPU.

Linux supports all this boards for 3d printers much better than Windows. No installation of drivers necessary. It is really Plug & Play.  

Regards
Gerald

macphyter, 2019-06-03:
If anyone has any doubts, I have an SKR 1.3 board, and I'm using it with Linux.  Works great!

gdachs, 2019-06-05:
@macphyter of course. In the meantime I got a SKR 1.3 too and Linux has no problems with it. But the first thing I did was to flash it with Klipper firmware.

anm8tor, 2019-09-04:
Any thoughts on why I am experiencing a problem with the SKR Mini then? I can connect via USB to my PC and talk to the board through Repetier Host, but when I connect to my Raspberry Pi 4, the Pi doesn't even recognize that something is plugged into the US port