bigtreetech/BIGTREETECH-SKR-V1.3 #2: Missing schematics

VanessaE, 2019-03-03:
The schematic and board trace layout files are missing....

jourjine, 2019-03-09:
Same request for schematic

sciencecentredelft, 2019-03-13:
Check this website: http://elsainmac.tistory.com/m/470
Translate it to English

VanessaE, 2019-03-13:
I see no schematics on that site.

zillarob, 2019-03-27:
This is what we are looking for, but for v1.3

[image]

bigtreetech, 2019-03-29:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/hardware/SKR-V1.3-SCH.pdf here is the SKR V1.3 schematic

VanessaE, 2019-03-29:
Yay!

bill-orange, 2020-02-14:
Schematic link no longer works.

VanessaE, 2020-02-14:
the moved the repo a bit, new link is https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/hardware/SKR-V1.3-SCH.pdf

VanessaE, 2020-02-14:
@bigtreetech  the schematics are still missing for the v1.4 though.

Coffee0297, 2020-02-14:
@bigtreetech can you please upload schematics for the 1.4? as soon as possible

GeorgesOatesLarsen, 2021-01-12:
This should be closed

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4-SCH.pdf

closed 2021-01-12 by VanessaE