bigtreetech/BIGTREETECH-SKR-V1.3 #3: Stepper driver diag0 and diag1 pins missing connectors.

drmulligan, 2019-03-04:
In your SKR v1.3 youtube video there are female header pins for TMC2130 diag0 and diag1.  On your aliexpress and github pages none of the diagrams show a connector for those two pins for each driver for sensorless end stops.  Are they present on the production boards you are selling or not?  If not will they work if I connect them?﻿

Msq001, 2019-03-28:
yeah, it has red female headers pins in the selling board.

Msq001, 2019-03-28:
Refer to this wriing diagram https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/Wiring%20diagram/SKR-V1.3-TMC2130-SPI-Mode.png

XtracT, 2019-04-22:
Hi

> Refer to this wriing diagram https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/Wiring%20diagram/SKR-V1.3-TMC2130-SPI-Mode.png

Hi @Msq001,

Can you explain a little further? In the schematic I do not see DIAG0 anywhere. How are the two header pins connected? Do they need to be shorted, or they actually need to be connected to the stepper driver?

The wiring diagram could be improved a little bit.

Thanks!

Msq001, 2019-04-24:
[image]
the red line in this picture have be wired in pcb, you need only insert the jumper,  the DIAG0 of driver will be wired with X-(endstop signal)

damienmg, 2019-05-15:
Hi,

I tested the board today and the diagram shown is a little incorrect. The upper pin (not the lower) is connected to the jumper that connect it back to the endstop connection.

I supposed this jumper is doing only that. If you have an ABL then you probably need to remove the ZST jumper. Even without ABL, I have seen many time that sensorless homing should not be used for the Z axis...

I could not understand where the lower DIAG pin is wired to, I guess this is safe to just solder 2 pins for DIAG  to all my 2130 v2 like they have in the 2130 v3? And then just use jumper to replace the endstop.

This would certainly help getting better documentation.

marcoaargemiro, 2019-10-27:
oLA, ESTOU COM ESSA DUVIDA COMO INTALAR O TMC2130 V3.0 NO SKR 1.3

marcoaargemiro, 2019-10-27:
rEALMENTE JÁ PESQUISEI O UNICO QUE ENCONTREI FOI VOCÊ ATE O MOMENTO

damienmg, 2019-10-27:
For reference for future reader of this bugs, this article contains my findings at the end: https://medium.com/@damien.martin.guillerez/skr-1-3-tmc2130-on-a-creality-ender-3-b4ec4abfdfd1?source=friends_link&sk=d12813db331b024817192ed1e3c24774