bigtreetech/BIGTREETECH-SKR-V1.3 #4: LCD 2004 Anet A8

antonvier, 2019-03-06:
Hi, I have a LCD 2004 that came with anet a8, this one here: https://www.aliexpress.com/store/product/2-PCS-Anet-2004-LCD-Smart-Display-Screen-Controller-Module -3D-Printer-Parts-for-RAMPS-1 / 3126008_32830879237.html? Spm = a2g03.12010615.8148356.1.35973f9e6FoSkr

Can you tell me how to connect to skr v1.3?

Thank you very much in advance

antonvier, 2019-03-06:
sorry, that link:  https://www.aliexpress.com/store/product/2-PCS-Anet-2004-LCD-Smart-Display-Screen-Controller-Module -3D-Printer-Parts-for-RAMPS-1/3126008_32830879237.html? Spm=a2g03.12010615.8148356.1.35973f9e6FoSkr

antonvier, 2019-03-06:
is needed remove the spaces in the link...

antonvier, 2019-05-11:
noone?

SISLANGER, 2019-05-18:
The pins are wired differently to normal LCD 2004, try and look up pin out for the ANET LCD 2004 and the pin out for a normal LCD 2004 and use Dupont leads as a means of testing.

alftrek, 2019-06-10:
Hello, have you managed to make it work? Would you have the Marlin scheme and configuration? Thank you

powertomato, 2019-07-23:
@alftrek @antonvier The SKR V1.3 uses a 3.3V MCU the Anet A8 display is 5V. While the board is 5V tolerant for digital input, the buttons of the display use analog reads, which are not 5V tolerant. Me and others have managed to make it work, but it requires a hardware modification of the display and changes to the marlin firmware, as it is not supported. If you don't know what you're doing **you can easily damage your board**!

Here is the link to the issue of the Marlin firmware: https://github.com/MarlinFirmware/Marlin/issues/14552