bigtreetech/BIGTREETECH-SKR-V1.3 #11: ESP3D - where to connect?

ArturNadolski, 2019-04-07:
Is there any other usable UART besides TFT/AUX-1? I need to connect ESP3D and my TFT/AUX-1 is already used by TFT35.

Valdas2000, 2019-04-18:
I have a similar issue with BIGTREETECH SKR V1.3. board 
Seems the issue can be solved using SPI<->UART board SC16IS750
I hope to create a Smoothieware module to get ESP3D interacting via SPI.

Generic idea is to utilize LCD EXP1/EXP2 SPI pins 
The only issue with that solution is ESP3D SD card reader. I am not sure if it can be utilized in a proper way.

Because I busy on my primary job development of the module expected to be long. I hope share beta version in a 40-50 days (can't spend more than 15 minutes for this project per day).

doggyfan, 2019-04-18:
Well you could use UART1 if you're not using TMC drivers.

RXD1 pin is on the Z driver pins under the driver chip - Z-RX pin.
TXD1 pin is on the E1 driver pins under the driver chip - E1-RX pin.

Not sure if the HAL is setup to use UART1, it ought to be though I think.

Valdas2000, 2019-04-18:
From what I get from the source code ESP3D modile talks to the SerialConsole object (as well as TFT35). 
Based on /src/libs/Kernel.h the only a single SerialConsole instance is allowed. So it does not matter which UART is used, the point is that there is only one SerialConsole.
In any case the code must be changed but I am not sure if it is better to keep 2 SerialConsoles instead of create a "SerialSPI" module talking to SerialConsole.

_src\libs\Kernel.h_
`        // These modules are available to all other modules
        SerialConsole*    serial;
`
_src\libs\Kernel.cpp_
`
    // serial first at fixed baud rate (DEFAULT_SERIAL_BAUD_RATE) so config can report errors to serial
    // Set to UART0, this will be changed to use the same UART as MRI if it's enabled
    this->serial = new SerialConsole(USBTX, USBRX, DEFAULT_SERIAL_BAUD_RATE);
`

ArturNadolski, 2019-04-18:
> Well you could use UART1 if you're not using TMC drivers.
> 
> RXD1 pin is on the Z driver pins under the driver chip - Z-RX pin.
> TXD1 pin is on the E1 driver pins under the driver chip - E1-RX pin.
> 
> Not sure if the HAL is setup to use UART1, it ought to be though I think.
> 
> I have my ESP8266 module on the TFT connector, but then I don't have a TFT.

I'm using 4x TMC2208. I will try E1 which is not used but I don't think it will work because there is only one serial console defined at Marlin.