bigtreetech/BIGTREETECH-SKR-V1.3 #12: SKR1.3 + smoothieware + lcd

sancrum, 2019-04-10:
Hi
I have RepRapDiscount full graphics smart display.
How can I use it with SKR + smoothiware?

Msq001, 2019-04-11:
The SPI interface of `RepRapDiscount full graphics smart display` in smoothieware must be hardware SPI, but it's software spi wiring in SKR V1.3,   so if you want use smoothieware, you can only use extra wiring to share the SPI of the SD card for LCD

ShohninDmitriy, 2019-04-18:
[image]
[image]

Filcanna, 2019-04-20:
So you’re saying we need a custom cable to connect RRDGLCD on SKR v 1.3?

sancrum, 2019-04-20:
Yep.
Thank you guys!

closed 2019-04-20 by sancrum

Filcanna, 2019-04-20:
Well. My 1.3 board arrived with Marlin, but no menus. So I’m trying LCD on SmoothieWare (it works PERFECTLY with its cables, twisted, on a Sbase I own), but with SKR board on display there is only light, other functionalities are fine. Well, I’ll try DuPont bifemale cables ASAP.