bigtreetech/BIGTREETECH-SKR-V1.3 #13: Max voltage for ABL and auto power pin?

tilkor, 2019-04-11:
Hello, 
I was looking under the pin group and I didn't see a pin specific for "auto power off after print", I am assuming that PS_ON_PIN?

I am using an inductive z probe and I wanted to know the max voltage the Z-min pin can handle?

Thanks!

Alissonverd, 2019-04-23:
@tilkor I'm trying to request a pin header to next version of this board. 
Maybe you could help with my request, we need show to @bigtreetech how it's important.

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/16#issue-435716653

doggyfan, 2019-04-24:
I sent them a message via AliExpress a while back about wasting 5 pins on the TxD to TMC2208, because the software serial code that is used in LPC MCU's in Marlin doesn't need the TxD pins, it works in single pin mode (RxD pin only) if you specify the same pin for RxD and TxD, but the people behind the SKR board design has not made this a possibility, though you can modify the board yourself, but so easily fixed if the SKY designers would listen, thus providing 5 free extra pins for the user to use as they so wish,.

LongLiveCHIEF, 2019-05-13:
I'm not familiar with non-Marlin firmware, but I'm wondering if it was designed this way for smoothieware or klipper?

tilkor, 2019-05-14:
yeah, I have maxed out the board and had to do some creative work around to get the PS_ON working. Since the chip can only put out something around .09w of power, I ended up using a low/high SSR that draws very low amps to trigger my larger SSR that handles power to everything. 

I would also post on the Bigtreetech facebook page.