bigtreetech/BIGTREETECH-SKR-V1.3 #15: SKR 1.3 with MKS2208 v1.1 TMC Connection Error (UART)

Brainc0re, 2019-04-18:
I got my board today. For testing, i connected one stepper (Z). according to the manual i removed all jumpers from the STEP/DIR headers and set them on the UART headers. But i always get a TMC CONNECTION ERROR on the display...

This is may current build:
https://drive.google.com/file/d/1NryYj-VFD_dcWFG64mShB-L8xsBiXHr1/view?usp=sharing

doggyfan, 2019-04-18:
I had this and found out the problem was today.

On the TMC2208 driver board you need to solder the PDN to the N.C pin next to it. The serial connection will then be correct.

[image]

[image]

Brainc0re, 2019-04-18:
Okay. I will try it in ~20 minutes. Thank you

Brainc0re, 2019-04-18:
Nope. Still "TMC Connection Error" on the screen

doggyfan, 2019-04-18:
Have you also soldered the UART pad option on your TMC2208 board ?

Brainc0re, 2019-04-18:
The board has separate jumpers for UART

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/SKR%20V1.3-Instructions.pdf Site 5/6

doggyfan, 2019-04-18:
I highly recommend using the protection diodes on each driver board to protect them from back EMF etc.

[https://www.aliexpress.com/item/5PCS-Ultra-Silent-Protector-Stepper-Motor-Driver-Motherboard-Plug-type-Silent-Stepper-For-3D-Printer-Driver/32864664427.html](url)

doggyfan, 2019-04-18:
Also, comment out this line on the configuration_adv.h file ..

`
//#define SOFTWARE_DRIVER_ENABLE
`

Brainc0re, 2019-04-18:
> Also, comment out this line in your configuration_adv.h file ..
> 
> `//#define SOFTWARE_DRIVER_ENABLE`

If i define that, error still there and i cant control the stepper via "move axis Z" anymore

PapeJelly, 2019-04-19:
Check out their instructional video which also includes TMC2208 setup

**https://www.youtube.com/watch?v=oaXfXkPYHpw**

Brainc0re, 2019-04-19:
> Check out their instructional video which also includes TMC2208 setup
> 
> [](**https://www.youtube.com/watch?v=oaXfXkPYHpw**)

With the help from Vanessa_E, author of [TMC2208 UART on BigTreeTech/BIQU SKR V1.1 and V1.3 Controllers](https://www.instructables.com/id/TMC2208-UART-on-BigTreeTechBIQU-SKR-V11-and-V13-Co/), we found out, that my MKS Drivers are not able to communicate via UART. I need some Fyetc v1.2 or TMC2208 drivers from Eryone.

closed 2019-04-19 by Brainc0re

doggyfan, 2019-04-20:
That's not true, I'm using them just fine with the extra soldered connection.

Brainc0re, 2019-04-20:
I dont have solderpads on my Stepper drivers and soldering the to pins in the middle has the same function as the jumper on the board.

[image]

doggyfan, 2019-04-20:
Can you not solder a thin wire from the chips UART pin to the N.C pin ?
It's a bit tiny but easy enough to do under a good magnifying glass.

Brainc0re, 2019-04-20:
Vanessa_E made this suggestion too. I have so many tools and stuff, but no magnification/microscope and such thin wire. According to the datasheet from trinamic, the solder pads are 0.25mm wide and have a pitch of 0.5mm. Thats too small for me for soldering. I will order some v1.2 which have UART.