bigtreetech/BIGTREETECH-SKR-V1.3 #17: TMC2208 [TMC CONNECTION ERROR]

lospo, 2019-04-26:
I configured marlin and SKR1.3 to work in uart mode but i have the same message on the display.
The driver are BIGTREETECH TMC2208 SKU:UUU001758
The board is SKR V1.3, i follow the instruction here: https://www.youtube.com/watch?v=oaXfXkPYHpw
But i have the same message on display. 

Attached a photo of my driver.

Here is the output of my M122:
`                X       Y       Z       E
Enabled         false   false   false   false
Set current     800     800     800     800
RMS current     1436    1436   1436     1436
MAX current     2025    2025    2025    2025
Run current     25/31   25/31   25/31   25/31
Hold current    12/31   12/31   12/31   12/31
CS actual       0/31   0/31     0/31    0/31
PWM scale   	0       0       0       0
vsense          0=.325  0=.325  0=.325  0=.325
stealthChop     false   false   false   false
msteps          256      256     256     256
tstep           0       0       0       0
pwm
threshold       0       0       0       0
[mm/s]          -       -       -       -
OT prewarn      false   false   false   false
off time        0       0       0       0
blank time      16      16      16      16
hysteresis
-end            -3      -3     -3       -3
-start          1       1       1       1
Stallguard thrs
DRVSTATUS       X       Y       Z       E
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
X       0x00:00:00:00    Bad response!
Y      0x00:00:00:00     Bad response!
Z       0x00:00:00:00    Bad response!
E       0x00:00:00:00    Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW
ok`
[image]
[image]

doggyfan, 2019-04-26:
Have you solder bridged the option pads on the TMC2208 driver boards to connect the UART pin on the TMC2208 to the UART driver board pin ?

lospo, 2019-04-26:
I made no solder, I place jumpers in the board and attached these drivers.
I have no idea of what to solder/connect

doggyfan, 2019-04-26:
Try soldering these two little pads together on each of your TMC2208 driver boards ..

[image]

lospo, 2019-04-26:
I asked also to the supplier, and he give me that image as reference for soldering two pins.
Are the same or i am confused?
[image]

doggyfan, 2019-04-26:
oh ok, go with what they say. If that doesn't work try the other two pads.

lospo, 2019-04-29:
Nope, i soldered these pads but the trouble still remain.
And now its difficoult for me de-solder and solder the other two pads.
Do you think i could solder every three pads?
I send photos, please tell me if you see something  wrong. 
[image]
[image]
[image]
[image]

doggyfan, 2019-04-29:
You soldered the pads BigTreeTech said to solder. Try soldering the pads I showed you instead.

pauleley, 2019-04-29:
You need to apply 12 or 24v for it to work

MrG1981, 2019-05-08:
I have the same issue
the jumper on the driver is soldered and works properly with FTDI serial to USB
but not on skr 1.3 board powered by USB or 12v from power supply

lospo, 2019-05-15:
Hi, i've soldered the pads together (because i have not good skill) and now it works!
Thanks to all of you!

closed 2019-05-15 by lospo

doggyfan, 2019-05-15:
BigTreeTech are telling people to solder the wrong two pads together :( .. it should be the other two.

stricktech, 2019-05-15:
hello, i have done the same thing, but i am getting tmc connection error.

i have soldered all 3 pads together, placed on skr 1.3 with the uart jumpers in place.
set firmware to TMC2208 and turned on TMCDEBUG.

i have my board and lcd at work right now, so cant test from 12 volt, powering from usb can give a tmc connection error?

lospo, 2019-05-15:
Yes, USB power is too low.
You need tre 12/24v power

stricktech, 2019-05-15:
Ok thanks, I'll reload the firmware and test it from my power supply when I get home.

Just to clarify something I wasent sure about or have heard differing information,

All jumpers should be removed besides uart?
(Black jumpers under driver, and red jumpers to the right side of board)

lospo, 2019-05-16:
> Ok thanks, I'll reload the firmware and test it from my power supply when I get home.
> 
> Just to clarify something I wasent sure about or have heard differing information,
> 
> All jumpers should be removed besides uart?
> (Black jumpers under driver, and red jumpers to the right side of board)

I removed all, and it works. :D

LongLiveCHIEF, 2019-06-27:
> BigTreeTech are telling people to solder the wrong two pads together 

No they're not. You can use either combination, or even solder all 3 pads together. You need to be applying the right voltage to the board in order for the the tmc2208 to allow current to flow.  The low voltage from your laptop/PC won't do the trick.

Peterg125, 2019-06-27:
Same exact problem. I'm replacing my Ender3's main board with SKR v1.3 from bigtreetech. Using 2208 drivers. Have pads soldered, have jumpers on the Uart pins on the board. connected to the 24v powersupply. 
Important note:
My board came with a damaged capacitor on E0, so I switched the pin out defintions for E0 and E1 in code and am simply using E1 port for my extruder. Otherwise everything else is stock.

MakerThinking, 2019-10-16:
Hello, I have the same problem, I have tried all the options but I still have a tmc connection error on the lcd I have skr board v1.3 and tmc2208 silent stepper
[image]
[image]
[image]

hesham1967, 2021-01-19:
I think this has to do with software problem otherwise why I have TMC connection problem with Cura altimaker and Ideamaker but when I use control maker, I don’t have any problem with TMC

D4N1ELPAT, 2024-03-14:
[image]
Me funciono solde los 2 pats juntos y cuando conecte la placa funcionaba igual lo cheque en el pronterface con el comando m122 y funciono muchas gracias a todos