bigtreetech/BIGTREETECH-SKR-V1.3 #18: [BUG] M190 ignored (critical)

cheesecheese56, 2019-04-28:
The following defect was described in the Marlin 2.0.x build that this one is built from (note link below):

https://github.com/MarlinFirmware/Marlin/issues/13287

The gcode M190 is used to make the gcode wait until the bed has reached the correct temperature. However, in this build, M190 is ignored, so the nozzle and heat bed begin heating at the same time. This can increase the draw on the power supply to above specifications (in my instance, the main fuse on the power switch blew at 4A).

I tested the issue by forcing M190 in the gcode and noticed that both tools began heating concurrently. That's when I found the defect had been documented in the main branch. I assume that this branch came before this defect was addressed.

I believe this is a potentially critical bug because it can cause damage to hardware.

doggyfan, 2019-04-29:
I wonder what printer you're using ? .. you're power supply is underrated, that needs fixing !

cheesecheese56, 2019-04-29:
This is a FT-5r2 printer with a 480w/20a/24v power supply, using a 300x300mm bed. There's a lot of juice to get the bed up to temp at 55C. If you run the math, this is just enough to blow a 4A fuse, if the hotend and bed heat concurrently.

Nevertheless, I've updated the latest Marlin 2.0 bugfix, and the M190 gcode is now working properly. This confirms that the aforementioned bug is present in this build (you can try it yourself by running a print and watching the temps in Octoprint or running gcode M105).

joymonkey, 2019-07-09:
This seems extremely important. Has there been any update from bigtreetech since this report?

carl1961, 2019-07-09:
BigTreeTech has merged  to Marlin, you can download from Marlin.