bigtreetech/BIGTREETECH-SKR-V1.3 #19: [BUG] BL Touch doesn't work with "#define FAST_PWM_FAN" enabled on BIGTREETECH SKR V1.3

DrNeis, 2019-04-30:
Description
If I activate "#define FAST_PWM_FAN" in your most recent version of Marlin 2.0 my BL Touch Probe stops working connected to my Board " BIGTREETECH SKR V1.3". The Probe shows blue LED always on and doesn't react to GCODE.
If i uncomment that line again everything works fine.

Steps to Reproduce
Activate "#define FAST_PWM_FAN"
Use the hardware mentioned above
Expected behavior:
Fan should be silent if running on reduced Speed AND BL Touch should still work

Actual behavior:
Fans are silent, but BL Touch stops working.

davidcgu, 2019-06-10:
Are u using auto_fan option?  which pin are you using for it?

pauleley, 2019-06-10:
i have 
#define E0_AUTO_FAN_PIN FAN1_PIN
and works good.
"#define FAST_PWM_FAN"
is for Use to change the FAST FAN PWM frequency ?

davidcgu, 2019-06-11:
The point is that I use a chimera, so I don't have this pin available.... I'm not using the SD card from the LCD maybe this gives some spare pin?

davidcgu, 2019-06-11:
In fact the SD card uses 5 pins on the exp2   
[image]
[image]

[image]

stray77, 2019-06-22:
@davidcgu , Any chance you could link to the source images you've posted, I'd like to see and have them assuming they are skr v1.3 specific.

davidcgu, 2019-06-22:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3

David Caparros

przemeklev, 2019-06-29:
I have similar problem with my skr but I don't use FAST_PWM_FAN. When power on my printer 3dtouch start and that's all
Send m280 p0 s10(120 90 ) and nothing

carl1961, 2019-06-30:
The bltouch clones never worked good for me, ( I have two ) they or also not near as accurate as the official BLTouch. I went back to the official BLTouch and there newest is better than ever.

carl1961, 2019-06-30:
The bltouch clones never worked good for me, ( I have two ) they or also not near as accurate as the official BLTouch. I went back to the official BLTouch and there newest is better than ever. But if you search, you can find the code to test the 3dtouch with a ardunio off the board. I am at work, but soon as I find that tester code at home I can post it.

pyunramura, 2019-08-24:
Can confirm the same behavior on most recent Marlin version using a 3D Touch. Uncommenting '#define FAST_PWM_FAN' causes the probe to stop responding to gcode commands. Will raise issue in main Marlin repo as well.