bigtreetech/BIGTREETECH-SKR-V1.3 #20: Board does not boot anymore

SirPlotalot, 2019-05-02:
Hello !

Everything went fine with my new skr 1.3 ( Marlin 2.X ( Ender 5 ) for about 20 prints.
Suddenly when i start my printer i get a constant beep tone and
a blue screen ( Stock Creality screen).
So i disconnected everything switched to 5v jumper and tried to recompile the firmware as usual via PlatformIO. 
I get the error "board not found".  ( But it shows up normally under windows 7 )
- Already tried another sd card, same problem.
- checked fuses
- pushed reset button , nothing happens, red Led is on all the time.

I am thankful for every hint, thank you in Advance !

ArturNadolski, 2019-05-02:
Did You check fuses?

SirPlotalot, 2019-05-02:
Hello !
As i wrote, yes. Both.

ArturNadolski, 2019-05-03:
Disconnect all connectors, remove stepstics and connect directly to PC (switch jumper to USB +5V).
If You can't connect via serial port try to use board UART directly (AUX-1) to PC or by OctoPrint/ESP3D.
If mainboard is OK You should get some information on console.

SirPlotalot, 2019-05-04:
Hello Artur,
connected the board via UARt to Octoprint. Trying to connect fails because no manual baudrate works, Auto baudrate brings up this : 

Connecting to: /dev/ttyAMA0Changing monitoring state from "Offline" to "Opening serial port"Connected to: Serial<id=0x6abad110, open=True>(port='/dev/ttyAMA0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitorStarting baud rate detection...Changing monitoring state from "Opening serial port" to "Detecting baudrate"Trying baudrate: 115200

archikkk, 2019-05-18:
Hello, did you have config for ender5? I got new skr1.3 but it wont bootup eith mine config, can you please give yours to try

SirPlotalot, 2019-05-20:
yes sure.

signorecello, 2019-10-13:
Hi, did you solve the problem? I'm having the same issue...

SirPlotalot, 2019-10-13:
No, had to buy a new board.... Sorry bro

jairov4, 2019-11-05:
looks like some boards have the EXP1 and EXP2 bad oriented, check the pinout to confirm.
In my case, I had to cut the slot space to allow my cable to connect correctly oriented.