bigtreetech/BIGTREETECH-SKR-V1.3 #21: TMC2208 X connection... Error: All LOW

pauleley, 2019-05-02:
TMC2208 X connection... Error: All LOW

But same Stepper will work in Y,Z or E. So Stepper ok.

X will work with TMC2130 in SPI mode or TMC in Standalone mode. Think it may be the X-UART jumper pins ?. How can i test. Have 12V and USB to Raspberry PI plugged in and powerd up ok.

SirPlotalot, 2019-05-04:
I have the same trouble for X only.But instead of All LOW i have ALL HIGH.
I havent changed anything, the problem occured over night at the last boot.

pauleley, 2019-05-05:
i have orderd a new board and 5 more TMC2208 to test if its the board or me lol.

codycato, 2019-06-07:
> i have orderd a new board and 5 more TMC2208 to test if its the board or me lol.

Did the new board and drivers work?

pauleley, 2019-06-10:
> > i have orderd a new board and 5 more TMC2208 to test if its the board or me lol.
> 
> Did the new board and drivers work?

Sorry haved not fitted yet. still useing the same TMC2208's in standalone mode lol.

stray77, 2019-06-22:
https://www.youtube.com/watch?v=duNHOPlh2Pg&t=688s
The only thing I'd change is from this video is
#define SERIAL_PORT_2 = 0
I haven't done 2208s but I have my 2130's in SPI running.  Consider the new TMC2209's, the goodness of 2130s in a cooler running and uart capable package.

Dilbert0815, 2019-07-26:
@pauleley 
You way want to check your 2208 step sticks how the TX/RX pins are routed on your sticks.
May be there is another 1k ohm on the stick getting in line with the 1k ohm on the SKR TX internally.
You would need a good meter with a resistance mode verifying wich step stick port is wired directly to the TMC pin.
The use of these ports of the step stick is vendor dependend. So it's not an SKR problem but compatibility with potential non-Bigtree sticks.
The difference of your XYZ channels may be a small variation in the serial resistores causing only one channel failing.

See my TMC2209 post for an option to fix this with a unusual jumper setting in the SPI jumper field.

bywebberen, 2019-11-24:
Same Issue here . BUT for all my 2208' s

>SENDING:M122
		X	Y	Z	E
Enabled		false	false	false	false
Set current	600	600	600	600
RMS current	1049	1049	1049	1049
MAX current	1479	1479	1479	1479
Run current	18/31	18/31	18/31	18/31
Hold current	9/31	9/31	9/31	9/31
CS actual	0/31	0/31	0/31	0/31
PWM scale	0	0	0	0
vsense		0=.325	0=.325	0=.325	0=.325
stealthChop	false	false	false	false
msteps		256	256	256	256
tstep		0	0	0	0
pwm
threshold
[mm/s]
OT prewarn	false	false	false	false
OT prewarn has
been triggered	false	false	false	false
off time	0	0	0	0
blank time	16	16	16	16
hysteresis
-end		-3	-3	-3	-3
-start		1	1	1	1
Stallguard thrs
DRVSTATUS	X	Y	Z	E
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0x00:00:00:00	 Bad response!
		Z	0x00:00:00:00	 Bad response!
		E	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW 

> 
Any guesses?? I have soldered the patch underneath the stepstick...