bigtreetech/BIGTREETECH-SKR-V1.3 #22: Analog or digital pins

pauleley, 2019-05-04:
I have the BIGTREETECH-SKR-V1.3 and would like to control some RGB LED lights from it on the printer. I'm looking at  Thomas Sanladerer setup on YouTube. But i cant see any Analog or digital pins on this board. can i use the End Stop+ pins if so how any help please.

SISLANGER, 2019-05-18:
I am trying the figure this one out also, I just can't work out the analogue pin number from it, as setting to P1_24 lets say does receive a signal

davidcgu, 2019-06-10:
Trying to figure out were to get a couple of PWM pint for auto fan and cabin light funtions, though that maybe enstops + pins were working but apparently not,any movement on this direction?

SISLANGER, 2019-06-10:
https://www.makenvape.uk/3d-printing/3d-printing-guides/skr-v1-3-neopixel-bltouch/

bustedparts, 2019-06-20:
Good luck getting any support on something like that. I have emailed them on issues and posted them on here with no response. This seems more like a beta .

davidcgu, 2019-06-21:
bustedparts, finnaly found that is only one available PWM at 5VDC dedicated for Bltouch, if you don't have bltouch you can use this, otherwise you can use XTZ+ as PWM with the inconvenient than works from 0 to 1.5volt only so will require some conversion.  At the end the conclusion it's easy, this is a cheap board and works fine however is like a beta, where many things are still not developed and won't be, the good option it's the PRO version much pore capable, powerful and open to development and future possibilities than v.13.....

bustedparts, 2019-06-21:
Awesome find! This is exactly what is needed .  Hopefully when I change the
config file the board will do as I ask. I have had problems with it not
doing what I ask of it. I have been changing the config than deleting the
firmware.cur file and replacing it with the firmware.bin file to rewrite
the firmware. Just changing the config and resetting the board doesn't seem
to work.

davidcgu, 2019-06-21:
in order to change some config you need to edit, recompile and reflash as has being ever the case.  From other side the good point is that with new 32 bit boards is just to copu the new complied file (.bin) on the sd card and restart the board.  If you are using octoprint you can reflash directly from it , from your laptop without moving from your side, all is done automatically.

bustedparts, 2019-06-21:
I am using this as a smoothie , not marlin. From my understanding you
should only need to change the config and reset. But I thought that
recompiling the firmware.bin file would be a sure thing. I have done this
several times trying to get the board to switch the laser output to 2.4 or
2.7 . But I no matter what I changed it to it would not change. I can see
what mosfet it was activating as it has a led when it is commanded up. So I
am hoping this will work.

davidcgu, 2019-06-21:
Sorry I don't know how smothieboard works, with marlin it takes the changes fine

bustedparts, 2019-06-21:
Well, I tried. I think there smoothie firmware side of this board is
broken. They also have not released the source code for it either. So I
redefined the pin and no dice. It stayed on the fan port. It wouldn't be so
bad if the port was regulated to 5 volts. But it is not. Sits at 24 volts
when power on.  I was thinking for a minute. Wait. Maybe I will just use
marlin. Well  the laser stuff is not that well supported with marlin 2.0.
Only to 1.1.9. So I think it is down to either a smoothie board or super
grbl board.

Projectile007, 2019-07-21:
Used pin 1_21 on EXP 1 with a 24V to 5V Buck Conv for the NeoPixels power. Compiled and NeoPixel is good to go...

GuiltyPixel, 2020-01-26:
Another option might be to use serial to talk to another microcontroller which controls functions like lights and ventilation. Custom codes and menu items could provide an interface. Mxxx R### G### B### W### for example, then your gcode can include instructions to automate things.