bigtreetech/BIGTREETECH-SKR-V1.3 #23: Smoothieware+TFT35 source code not awailable

marko-kukovec, 2019-05-10:
I noticed there is source code for Marlin (https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/firmware/Marlin-bugfix-2.0.x), but there is no source code for Smoothieware+TFT35 (https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/firmware/smoothieware%2BTFT35). Just binary **firmware.bin** is available. As this code is based on the open source **Smoothieware** code (https://github.com/Smoothieware/Smoothieware), it should also be available as open source with all the needed modifications.

doggyfan, 2019-05-10:
Good luck in getting the Chinese to respect such agreements around the rest of the world ;)

CCoffie, 2019-05-25:
On a similar note if anyone has been able to get a more recent version of smoothieware running on the board that would be awesome.

bustedparts, 2019-05-31:
> On a similar note if anyone has been able to get a more recent version of smoothieware running on the board that would be awesome.
I am trying to get smoothie working on this board right now. It seems to load fine. But it does not respond to any g code commands at all.

macphyter, 2019-06-03:
I would also like to see the Smoothieware source code posted, I could merge changes into the upstream tree so that we can run the latest code.