bigtreetech/BIGTREETECH-SKR-V1.3 #24: Board doesn't boot after firmware install

codiac2600, 2019-05-11:
I've compiled and installed a known working Marlin 2.0 firmware and everything works great. 

I made my own Marlin 2.0 firmware for my custom printer and the SD card immediately doesn't work again. Try the other one and it works. 

I thought I was smart so I edited the working firmware and again it doesn't work. 

Here are my configs if anyone wants to take a peak: https://drive.google.com/open?id=12n-i3yIjOXyX-WQ-DNjE4RwaVvA7YZh3

Not sure what's up but I'v never had this issue with the MKS Gen L

xthulu, 2019-05-18:
I had this problem when I tried to use pid to control my bed temp (#define pidtempbed I think).  The  skr roots and quickly crashes before I can connect to it via usb.

Currently solved by commenting that out, but now I have other problems as yet unresolved.

marclefevre, 2019-07-31:
How can you tell if the board boots?  Is there an LED that should be active?  I can update firmware to the latest Marlin 2.0 bugfix and the board changes it to FIRMWARE.CUR, but it won't respond to Octoprint's first M110 command after reset.  Is there a serial port for debug that I can connect to?

marclefevre, 2019-07-31:
Disabling PIDTEMPBED made no difference for me.

pmozbert, 2019-08-31:
I had this same problem when using the SD card that came with the board.  In the end, replacing the SD card fixed the problem.