bigtreetech/BIGTREETECH-SKR-V1.3 #25: TMC Connection Error

stricktech, 2019-05-15:
lcd shows TMC CONNECTION ERROR , and if i send m122 i get all low on xyz and e.

i have soldered the j2 pads, set TMC2208 , set both serial ports as shown in the instructions in config, set tmcdebug in config adv. 
I have all jumpers removed except the uart jumpers.

is there something im missing?

SISLANGER, 2019-05-18:
Just checking you set the firmware as in the video

https://www.youtube.com/watch?v=oaXfXkPYHpw

Javi91ac, 2019-05-20:
Try to solder 3 pads j2. So I fix it

Yannick06300Pat, 2019-05-28:
Alors oui souder sur j2 2/3 ou 1/2/3
Réussi avec tmc2208 v3
[image]

SISLANGER, 2019-05-28:
V3 Doesn't need soldering if you got the ready soldered version

if need to solder
https://www.makenvape.uk/3d-printing/3d-printing-guides/skr-v1-3-setup-guide/#TMC2208_SOLDER

stricktech, 2019-05-28:
Sorry, should hahavhahaveclosed this long ago, its bc I was powered with usb at the time.

closed 2019-05-28 by stricktech