bigtreetech/BIGTREETECH-SKR-V1.3 #28: BLTOUCH configuration issue

jraSA, 2019-05-23:
I've followed the instruction of this video to configure my BLTOUCH sensor:

https://youtu.be/oaXfXkPYHpw?t=244

But that seems to be outdated, the video says to uncomment this line :
`#define BLTOUCH`
and then comment this one:
`\\#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`
But after that it says to uncomment this line, that no longer exists:
`#define Z_MIN_PROBE_ENDSTOP`
I've tried skipping this step but as results my BLTouch doesn't work and stay blinking with red lights.
I'm missing something?, I've checked the pin connections and everything seems to be ok

LongLiveCHIEF, 2019-06-27:
@jraSA have you figured this one out yet? I'm setting mine up today. With the version of the firmware that's current as I write this, the firmware is set for the BLTOUCH to be in 3.3 V mode by default. According to the BLTouch docs, you have to cut a trace on the BLTOUCH board itself in order for 3V mode to work.

I believe the SKR-v1.3 is 5V tolerant, and if so, that means that in `configuration_adv.h` you should be able to uncomment `#define BLTOUCH_5V_MODE`.

I'm double-checking this as we speak, but if true, then enabling 5V mode in the firmware should do the trick without needing to modify the BLTouch itself.

LongLiveCHIEF, 2019-06-27:
confirmed.  set `#define BLTOUCH_5V_MODE` to get it to work. 

`Z_MIND_PROBE_ENDSTOP` was actually redundant, and was removed via https://github.com/MarlinFirmware/Marlin/pull/13276

bartofborg, 2019-06-28:
> confirmed. set `#define BLTOUCH_5V_MODE` to get it to work. Pins for servo are `+5V`, and the source pin for `X_ENDSTOP` is also `+5V`. Make sure you're plugging the White lead from the BLTOUCH into the Source pin of the endstop, and the black lead to the Gate (it goes VGS from top to bottom).
> 
> For the servo pins, you want the yellow lead from BLTOUCH on the far right, the Red lead on the far left, and the brown lead in the middle.
> 
> `Z_MIND_PROBE_ENDSTOP` was actually redundant, and was removed via [MarlinFirmware/Marlin#13276](https://github.com/MarlinFirmware/Marlin/pull/13276)

I am also just setting up my Wanhao I3 with this board and I cannot find any "#define BLTOUCH_5V_MODE"in config.adv.h?

LongLiveCHIEF, 2019-06-28:
Are you using the latest tip of Marlin bugfix-2.0.x?

I have a set of working configs for the latest bugfix-2.0 with the skr-v1.3 and BLTOUCH here: https://github.com/HackerHappyHour/Marlin/tree/ender3-skr-v1.3-config

bigtreetech's configs are months old, and have many errors to boot. I wouldn't use them until they update them.

I may be submitting some patches to their configs as well as the official Marlin project upstream.

Until then, use my configs.  The link I gave you is guaranteed to be stable, as I will create test branches to test config changes, and treat the branch I linked you to as a production branch.

Use this comparison link to see what settings I changed from the default marlin configs, and make sure to change CHOPPER voltage to 12V if your printer uses a 12V supply.

Also note that I use the Z_MIN_ENDSTOP pin instead of the Z_MAX_ENDSTOP pin, which is the only connection cable difference from bigtreetech's youtube video.

compare link: https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...HackerHappyHour:ender3-skr-v1.3-config

LongLiveCHIEF, 2019-06-28:
related #40

crencirs, 2019-07-11:
Hi,

I had a similar problem. Can you change the brown and red lines?
I had RAMSP 1.4, which is in the order of Brown(-), Red (+), Yellow(S) for servo PIN.
But the SKR 1.3 board is in the order of Red(+), Brown(-), Yellow(S).

So I had to change the order of the lines.