bigtreetech/BIGTREETECH-SKR-V1.3 #29: macOS support

ganzocrypt, 2019-05-27:
Hello,

I ordered 2 boards and wanted to know if there is macOS support for connecting to the board.
If no, any plans?
thx

jarjartee, 2019-06-01:
Appears  NO support from BigTreeTech - having a NIGHTMARE trying to get thing set up - beginning to REGRET purchase

biskero, 2019-06-01:
I joined their Facebook page, will ask there too! Hopefully there is a solution

LongLiveCHIEF, 2019-06-27:
Board makers don't support operating systems. Operating Systems support communication protocols needed to connect to boards. This is a motherboard for a printer, and it's basically the same thing as an arduino, teensy, etc...

Whatever issues you have connecting to Mac are going to be the same as any other embedded device. You need to make sure your mac has drivers for serial communication, and that your mac user has permissions to the groups that own usb serial device connections.

Just google "connecting arduino to Mac" and follow any solutions/instructions and they will work for this device also.

biskero, 2019-06-27:
I have no problem with the board and mac. Others reported issues with mac. In any case a month has passed .....

LongLiveCHIEF, 2019-06-27:
yeah, they're really bad about responding to these issues. I want to submit a PR for the whole BLTOUCH thing #28, but I'd have to manage 4 months of updates to the configuration and configuration_adv files along with this merge and that's just not worth it.

If you're interested, I've got my own firmware setup for this board tracking the latest bugfix-2.0.x in https://github.com/HackerHappyHour/Marlin/tree/ender3-skr-v1.3-config.  You can run a compare with the head of Marlin's own branch to quickly see what I've changed to get things to work for me using this link: https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...HackerHappyHour:ender3-skr-v1.3-config

biskero, 2019-06-27:
I will take a look! I got a very custom huge printer to setup, just waiting for new 2209 drivers so I do not have to deal with strange configs for 2130/2208 ! thx