bigtreetech/BIGTREETECH-SKR-V1.3 #32: SKR v1.3 PS_ON Pin

djice024, 2019-05-28:
I am building my own 3d printer, it is a delta printer so I decided to go for a 32-bit board. After some research, I concluded that the SKR 1.3 was the best option. With only one problem. I want to use an ATX power supply but can't find a PS_ON# pin on the SKR 1.3. Is there any way I can assign a PS_ON# pin or cant this board do this? Thanks for the help in advance.

doggyfan, 2019-05-28:
Try adding (Y_MAX pin as an example) ..

```cpp
#define PS_ON_PIN    P1_26
```

it ought to work OK.

The SKR 1.3 board is kind of short of pins, which is probably why no one has added the pin (don't know). 
But BigTreeTech could have made more physical pins available. For a start, the CPU has 2 pins which aren't used at all (P1.0 and P0.26), you could solder to them if you're able. Then there are the 5 software UART Tx pins that could have had a pin header for the user to use as they so wish. And then they could also have put a pin header for the SPI motor lines (MOSI, MISO and SCK), though you can get to them under the driver boards.

djice024, 2019-05-29:
Yeah, the SKR 1.3 does have some flaws. I still feel it is better than just a RAMPS with a 32-bit controller. Doesn't the PS_ON need a connection to ground to turn on? And where will I be able to input the 5v? Thanks for the help.

doggyfan, 2019-05-29:
Not sure what you mean PS_ON needs connection to ground ?
The PS_ON pin will be an output pin from the board, How you connect it to your PSU will depend on the PSU's needs.

To feed +5V to the board, see the attached image ..

[image]

djice024, 2019-05-29:
The transistor would be between the PSU's PS_On pin and ground, controlled by the SKR 1.3, Right? the power makes sense. Thanks a lot.

doggyfan, 2019-05-29:
You've got three options really ..

[image]

A bit of an over-kill but you could just buy a little board to go between such as ..

https://www.ebay.co.uk/itm/3-20V-Mosfet-MOS-Transistor-Trigger-Switch-Driver-Board-PWM-Control-Module-CDUK/382886645254

djice024, 2019-05-29:
I'll try that out, Thanks.

Nonametruck, 2019-06-02:
If you are using the MKS tft there are pins there right?

djice024, 2019-06-02:
That's for the MKS PWC and you'll still need a PS_ON pin. This was for an ATX power supply.

djice024, 2019-08-25:
> Here's five possible options (their are more) ..
> 
> [image]
> 
> A bit of an over-kill but you could just buy a little board to go between such as (#define POWER_SUPPLY 2) ..
> 
> https://www.ebay.co.uk/itm/3-20V-Mosfet-MOS-Transistor-Trigger-Switch-Driver-Board-PWM-Control-Module-CDUK/382886645254

I have been doing some experimenting. I ended up using an NPN transistor. I tried using the resistor config you posted, but it didn't seem to work with that. It did work without the pullup resistor so I assume the board has a built-in one. I ended up using the BLtouch pin. I wanted to thank you as you helped out an immense amount. I attached of my pins.h and a pic of the printer.
[image]
[image]

closed 2019-08-25 by djice024

leonyd0823, 2019-08-26:
Buenos días estoy tratando de poner unas tiras LED en esta placa skr v1.3 pero en realidad no se de donde tomar los pines ni cuales serían los más adecuado. Tanto para una tira LED RGB que una tira LED digital. La verdad busco un poco de información sobre este tema y un poco de ayuda muchas gracias el objetivo es poder controlar las tiras LED con la placa según el estado de la impresión.

djice024, 2019-08-26:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/63

djice024, 2019-08-26:
> Buenos días estoy tratando de poner unas tiras LED en esta placa skr v1.3 pero en realidad no se de donde tomar los pines ni cuales serían los más adecuado. Tanto para una tira LED RGB que una tira [...]

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/63

oxcxsxixbxaxn, 2019-09-16:
> > Here's five possible options (their are more) ..
> > [image]
> > A bit of an over-kill but you could just buy a little board to go between such as (#define POWER_SUPPLY 2) ..
> > https://www.ebay.co.uk/itm/3-20V-Mosfet-MOS-Transistor-Trigger-Switch-Driver-Board-PWM-Control-Module-CDUK/382886645254
> 
> I have been doing some experimenting. I ended up using an NPN transistor. I tried using the resistor config you posted, but it didn't seem to work with that. It did work without the pullup resistor so I assume the board has a built-in one. I ended up using the BLtouch pin. I wanted to thank you as you helped out an immense amount. I attached my pins.h
> [pins_BIGTREE_SKR_V1.3.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/3538219/pins_BIGTREE_SKR_V1.3.zip)

@djice024
So which of the 5 scenarios did you use?  I have a the components handy, but did you use the NPN only? and neither of the resistors shown?  
Oh, the NPN is BC547, I think that is still appropriate, right? 
Thx