bigtreetech/BIGTREETECH-SKR-V1.3 #33: NO MacOS support ??

jarjartee, 2019-05-29:
So I was happy to receive my SKR  1.3 only to realise that there is little or no support for Mac OS. I know the firmware is based on Marlin 2.0x  but  disappointed to see little or no activity in responding to issues raised her on github.  I have tried to use both the original BigTreeTech firmware and also the latest Marlin 2.0 release - flashing  it to the board / SD card.  I have had very little success in communicating with the board via USB cable (from my Mac) also octopi will not connect or communicate with the board. I am hoping that BigTreeTech will respond to the issues raised her on  GitHub very soon

jarjartee, 2019-06-02:
I’ve got to draw a close to trying to get my SKR 1.3 working - and trust me I’ve tried ANYTHING and EVERYTHING , can’t justify spending anymore time on it. Opened Dispute via AliExpress and Had someone from BiQU assisting via RemoteDesktop (MAC) but to no avail. I suspect hardware failure. What bothers me is the randomness of ‘non working boards reports‘ here and other forums - also NO updates to Firmware from BigTreeTech since launch. Marlin 2.0 repo is clearly active in big quashing so I’m hoping to see more CONCRETE examples of working boards (there are many). New ‘bleeding edge tech’ will always be tricky at first but I’m not sure if the SKR 1.3 was quite ready for release - my experience is obviously tainted. I’m hoping for a replacement board once my dispute is resolved and in between time some more issues resolved in the BigTreeTech repo.t

LongLiveCHIEF, 2019-06-27:
I'm sorry you're having these problems, but you should realize that with embedded systems, it's a matter of the OS supporting the communications, and not the other way around.

Board makers don't get to choose which operating systems they do or don't support. 

The most likely reason you're having communications issues is that your mac is missing a driver for serial communications, or your primary user isn't a member of admin/wheel groups that are used by the ttyUSB device mapper on your Mac.

ciscomonkey, 2019-07-10:
I'm mac here and got two SKRs that are working with it.  I use the instructions on the Marlin site to install the plugins for Sublime Text 3 and works great.  I can connect up to the boards with pronterface and using serial console just fine so far.

wabbitguy, 2019-07-14:
@ciscomonkey in Marlin 1.1.9 when I connected with Pronterface I'd get a bunch of printer settings echo'd back to me on connection. With SKR I get a single line that says "Printer Online". I can send GCODES and it responds but I don't get the usual verbiage that I did before.

Just wondering if that's normal or something new in Marlin 2.0?

Mel

ciscomonkey, 2019-07-15:
Normal for me here. With old Marlin when you connect it reloaded the mega
which is why you would probably see that. When you connect the skr it
doesn't do that. You can send the g codes to get the info still though.

wabbitguy, 2019-07-16:
@ciscomonkey thank you very much for the clarification and the reason why it did that!

Very much appreciated...

Mel

Ohmarinus, 2019-10-10:
Sadly, I am also unable to get the SKR v1.3 working on mac. Both of the USB ports are not recognizing the SKR v1.3. I am clueless as to what could be the solution. I can upload the file to the SD-card but I need to be able to use my computer as a host so that I can control and configure the motherboard. In other words, the SKR is not recommended when using a Mac I guess. I wish they would've told me before I bought the product.

Ohmarinus, 2019-10-10:
> So I was happy to receive my SKR 1.3 only to realise that there is little or no support for Mac OS. I know the firmware is based on Marlin 2.0x but disappointed to see little or no activity in [...]

If you cannot get it to show up, try moving the 5v jumper pin from INT to USB. I could also not get it to work, but after moving the jumper I managed to get it to mount on my mac without modification. I believe this is the error that most people run into when using the SKR board so as a last resort you can try this.

LongLiveCHIEF, 2019-10-10:
@wabbitguy that's not a workaround, that's the real setup process.  That's also how you're going to be updating the firmware each time.

There really wouldn't be an inherent increase in print quality just by switching to a 32-bit board.  The main reasons you _might_ see improvement as a result of switching would be related to your use of a print server (if you use one) and the lag in communication between the print server or lack of bandwidth between the two resulting in skipped gcode.

The real benefits for a 32bit upgrade are the ability to handle more features in the firmware while being able to maintain gcode processing speed.

If you were using Marlin 1.x, and printing gcode from an sdcard, you won't see any changes in print quality just by upgrading.

Ohmarinus, 2019-10-10:
When you use a machine like a CoreXY or a Delta you will usually see improvements with a 32-bit board. In all other cases, most likely not. For a Delta the 32-bit board is the most useful. The other cases being when you use 128 to 256 microstepping drivers.

wabbitguy, 2019-10-10:
I am using the SKR on two deltas. No difference that I can detect other than the higher speed capability.

I use TMC2208 drivers but not programmed in UART style, just legacy and drop in replacements for the 4988's.