bigtreetech/BIGTREETECH-SKR-V1.3 #35: [BUG] No TMC2208 connection on SKR v1.3

KoffeinKaio, 2019-06-12:
### Description

After removing the jumpers under the Steppers and inserting the UART jumper, TMC2208 is still recognized as "ALL LOW"

### Steps to Reproduce

1: Get TMC2208
2: Solder bridge for UART to PDM pin
3: remove jumper between stepper pins on the SKR2.3 board (not sure about that)
4: insert jumper on UART pins for axis
5: insert stepper driver
6: boot up (assuming all configuration in FW is set)

**Expected behavior:** [What you expect to happen]
UART Connection made

**Actual behavior:** [What actually happens]
UART Connection ALL LOW

#### Additional Information

FW Configuration:
@platformio.ini: env_default = LPC1768
```
  #define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3 
  #define X_DRIVER_TYPE  TMC2208
  #define Y_DRIVER_TYPE  TMC2208
  #define Z_DRIVER_TYPE  TMC2208
```

Picture of stepper:
https://frupic.frubar.net/shots/38178.jpg

Picture of board:
https://frupic.frubar.net/shots/38177.jpg

DJDOUBLEG, 2019-06-20:
TMC uart mode work only when connect the power supply. Only with USB not work.

MrG1981, 2019-07-11:
Hi you have the same drivers as me
You Need to solder the pins NC and PDN together

Then it works

Evg33, 2019-07-11:
> Only with USB power

Only with USB power drivers not powered and cannot be reconfigured by uart interface.
It is on any board.
Use PSU power or M502 and M501 for init drivers in startup code before printing.

MrG1981, 2019-07-11:
Yes you right @Evg33
But this drivers don't work without the extra solder connection
You can power them with psu but the uart wouldn't work

KoffeinKaio, 2019-07-11:
The problem was indeed the not-connected pins, thank you all

closed 2019-07-11 by KoffeinKaio