bigtreetech/BIGTREETECH-SKR-V1.3 #36: Smoothieware not changing ports after defining.

bustedparts, 2019-06-16:
I have been working on getting this board to work on a K40 laser. When I define the #laser_module_pin  to 2.4 . The firmware doesn't seem to respond. When it is commanded to fire through software, the 2.3 mosfet led lights up.  So I tried 3 times with 2 different boards (which I may add both ended up breaking) to change the pwm pin . Now through what I have read there is alot of pins that can be used for this function. But without the board doing what it is told it is useless. So I am at a loss. Is there a dev that can help me. I was going to post this setup on several sites to promote it. So please lend a hand!

bustedparts, 2019-06-20:
After receiving a replacement board I have been doing testing with smoothie. When you define laser to true, it seems to default to pin 2.3 . And defining anything else does nothing. It will not change from that port. Further more pin 2,3 on the "-" pin it will enable the laser to cut but there is no regulation.

BlakeEnterprize, 2019-11-10:
how did you install the stepper drivers?