bigtreetech/BIGTREETECH-SKR-V1.3 #37: Thermocouple board connection

targor, 2019-06-18:
To which pins do i need to connect thermocouple boards and do i need to set other settings in the firmware except the ones below?

 *    -4 : thermocouple with AD8495
 *    -3 : thermocouple with MAX31855 (only for sensor 0)
 *    -2 : thermocouple with MAX6675 (only for sensor 0)
 *    -1 : thermocouple with AD595

Some pictures with the contacts to be used marked on the board would be nice.

stray77, 2019-06-21:
[https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/Wiring%20diagram/SKR%20V1.3%20Wiring%20diagram.jpg](url)

stray77, 2019-06-21:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/Wiring%20diagram/SKR%20V1.3%20Wiring%20diagram.jpg?raw=true

targor, 2019-06-21:
These are connections for thermistors not thermocouples. I want to know how i can connect boards (e.g. [this one](https://www.aliexpress.com/item/MAX31855-K-Type-Thermocouple-Breakout-Board-Temperature-Measurement-Module-For-Arduno-Wholesale-price/32920349240.html?spm=2114.search0104.3.1.7d0b5f0fS9Kcg5&ws_ab_test=searchweb0_0,searchweb201602_10_10065_10068_10547_319_317_10548_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10843_10059_10884_10887_321_322_10103,searchweb201603_52,ppcSwitch_0&algo_expid=ca323790-379f-41fd-a939-2fbaab7ff5af-0&algo_pvid=ca323790-379f-41fd-a939-2fbaab7ff5af&transAbTest=ae803_4) ) for thermocouples, and what kind of changes i must take in the firmware to run them. If anyone could give me some help here i would be very happy :)

stray77, 2019-06-21:
Is this helpful? https://www.instructables.com/id/How-to-use-MAX6675-Thermocouple-Amplifier-with-RAM/
[image]

osseum3d, 2020-09-12:
Hi, I'm trrying to add a Adafruit MAX31855 to the SPI port on Motherboard. But I get and error of "no printer connected". Could you help me?