bigtreetech/BIGTREETECH-SKR-V1.3 #42: BigTreeTech SKR board.stl is wrong

pw-sys, 2019-06-29:
BigTreeTech SKR board.stl is wrong the outer connector of the stepper driver is 2.4mm too far in.
The two big connectors are (heatbed and 24Vin) are 1.3mm to wide
TH1 TH0 TB are 1.3mm too far to the right and should be touching AUX-1 and should be 23.9mm wide not 22.7mm.

USB Port is 1mm to narrow 

[image]
[image]

carl1961, 2019-06-29:
Most likely there Board had been revised since they did that STL and also why would you need that STL I think it's just a visual look

LongLiveCHIEF, 2019-06-30:
The STL helps to model enclosures or other extensions to the board

LongLiveCHIEF, 2019-06-30:
@pw-sys does this thigiverse thing have it right? https://www.thingiverse.com/thing:3701548