bigtreetech/BIGTREETECH-SKR-V1.3 #43: Stepper Extender on SKR v1.3

yoch2015, 2019-06-30:
I assigned an end stop pin because I wanted to set E to 3, but it does not work well.
Do you have any advice?
[pins_BIGTREE_SKR_V1.3.h]
#define E2_STEP_PIN        P1_27
#define E2_DIR_PIN         P1_29
#define E2_ENABLE_PIN      P1_04
#ifndef E2_CS_PIN
  #define E2_CS_PIN        -1
#endif

carl1961, 2019-06-30:
haven't ventured that far yet, but maybe this video will help

https://www.youtube.com/watch?v=p6bAMM-XQL0

yoch2015, 2019-07-01:
Thank you carl1961.
Because I use a diamond hot end, I needed three extruders.
Looking at the schematic of SKR v1.3, we decided that the end stop pin is not appropriate because it has passive components connected, and tried to assign another pin, it reports as it was successful.
The stepper motor moved with E2 active with the following settings and connections. As a bad effect, E0 will also be held during E2 operation.

**[pins_BIGTREE_SKR_V1.3.h]**
#define E0_STEP_PIN        P2_13
#define E0_DIR_PIN         P0_11
#define E0_ENABLE_PIN      P2_12
#ifndef E0_CS_PIN
//  #define E0_CS_PIN        P1_08
  #define E0_CS_PIN        -1
#endif

#define E1_STEP_PIN        P0_01
#define E1_DIR_PIN         P0_00
#define E1_ENABLE_PIN      P0_10
#ifndef E1_CS_PIN
//  #define E1_CS_PIN        P1_01
  #define E1_CS_PIN        -1
#endif

#define E2_STEP_PIN        P1_08
#define E2_DIR_PIN         P1_01
#define E2_ENABLE_PIN      P4_28
#ifndef E2_CS_PIN
  #define E2_CS_PIN        -1
#endif

[image]
[image]

closed 2019-07-01 by yoch2015