bigtreetech/BIGTREETECH-SKR-V1.3 #44: DRV8825 loaded but wont move steppers

daimein2007, 2019-07-01:
I just received the board, installed the firmware, installed correct pins and installed DVR8825 drivers but I get no movement from the stepper motors, not even a slight resistance when turned by hand. I set voltage 0.8 the driver chips are getting hot just don't seem to be working. I rechecked firmware 4 times using atom firmware compiled and uploaded just fine. I'm
[Marlinconfighandadv.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/3347384/Marlinconfighandadv.zip)

 stumped.

pauleley, 2019-07-02:
have you put the jumpers on under the drivers. on MSI 1 and 2

daimein2007, 2019-07-02:
[image]

Just like this is how i did it

xoniq, 2019-07-02:
In had a different problem. Drivers were working, but I got a huge amount of missing steps, every single print got skewed majorly.

First I blamed my belts for being not properly tensioned. Didn’t help, interlaced my belts and motors, the same. Reused my Ramps for testing, everything working fine.

After buying 4 TMC2130 drivers, in SPI mode, everything is rock solid on the SKR. So it seems that my SKR doesn’t like my DRV drivers which worked perfectly fine on ramps. So I keep the ramps with DRV drivers aside as spare or for a second printer or so.

My suggestion, to fully use all the advantages of wireless SPI on the SKR, invest in TMC2130’s. Bonus; sensorless homing.