bigtreetech/BIGTREETECH-SKR-V1.3 #46: ((TMC connection Error))

Salim169, 2019-07-06:
Dear Friends ,
Since 2 week i work on SKR 1.3 with BLTouach and TFT SKR.
and i have The Original TMC 2130 from WATTEROTT .
when i configured the board with Marlin-bugfix-2.0.x .
i have always ((TMC connection Error)) on / display 12864/, 
and on Pronterface With (( M122 Code)) i have:

SENDING:M122 S0
		X	Y	Z	E
Enabled		false	false	false	false
Set current	800	800	800	800
RMS current	1436	1436	1436	1436
MAX current	2025	2025	2025	2025
Run current	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31
CS actual		0/31	0/31	0/31	0/31
PWM scale	0	0	0	0
vsense		0=.325	0=.325	0=.325	0=.325
stealthChop	false	false	false	false
msteps		256	256	256	256
tstep		0	0	0	0
pwm
threshold		0	0	0	0
[mm/s]		-	-	-	-
OT prewarn	false	false	false	false
off time		0	0	0	0
blank time	16	16	16	16
hysteresis
-end		-3	-3	-3	-3
-start		1	1	1	1
Stallguard thrs	2	2	0	0
DRVSTATUS	X	Y	Z	E
stallguard
sg_result		0	0	0	0
fsactive
stst
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0x00:00:00:00	 Bad response!
		Z	0x00:00:00:00	 Bad response!
		E	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW

my mode is: TMC 2130 /SPI Mode/Sensor less homing for X and Y BLTouch. 
Of course, i changed the processor name to(LPC1768) and motherboard name to (BOARD_BIGTREE_SKR_V1_3) 

So I need your help and suggestions Please

xoniq, 2019-07-06:
Tried with 12V on the board? (Jumper not to USB) otherwise they won’t show correct status.

So jumper to INT, 12v on the board, and then connect to pronterface

Salim169, 2019-07-06:
[image]
[image]

Salim169, 2019-07-06:
the jumper already on the int+5V .
and my power supply is 24V

xoniq, 2019-07-06:
SW SPI enabled? Is the TMC driver itself set to SPI physically?

Ghenghis, 2019-07-07:
I have the same Issue using TMC5160
SENDING:M569
X driver mode:	spreadCycle
Y driver mode:	stealthChop
Z driver mode:	stealthChop
E driver mode:	stealthChop

m122
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0xFF:FF:FF:FF	 Bad response!
		Z	0xFF:FF:FF:FF	 Bad response!
		E	0xFF:FF:FF:FF	 Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All HIGH
Testing Z connection... Error: All HIGH
Testing E connection... Error: All HIGH

Salim169, 2019-07-07:
@Ghenghis  and do you use your Printer with this Errors?

Ghenghis, 2019-07-08:
no last night I switched back to my old board. no clue whats wrong. have couple other boards coming so will try again.

mpallen509, 2019-07-08:
I found a few issues with my first board from Bigtreetech. Hopefully testing the new boards will result in a solution. 

If the problem persists, consider attaching your Configuration.h and Configuration_Adv.h

Salim169, 2019-07-09:
@Ghenghis @mpallen509  I  I have concluded that the painting does not work.
I wait also  my new Ordr(SKR 1.3 +Bigtree Tech TMS  2130 SPI) .

//Keep us connected in the results//

Ghenghis, 2019-07-13:
I had to remove CLK Pin on the tmc5160 v1.1 & #define MINIMUM_STEPPER_PULSE 1
I was able to move Motors after this.

Mopa92, 2019-07-18:
Hi,
I've the same problem, SKR 1.3 & 4 x TMC2208 from bigtreetech in UART mode. TMC error and testing XXX connection...Error ALL LOW.
In the other hand, the printer can print without any problem...

Salim169, 2019-07-20:
i received today my new  SKR 1.3 with TMC2130 .
 and i will see what i got......

Salim169, 2019-07-23:
After several experiments onSKR1.3 board ,with Original  TMC 2130 (SPI mode)  from WATTEROTT, my problem was : i had always message on the display ((TMC connection Error)),  and ended the first experiment with board loss ended  .

But everything was done successfully with the second board and this time with Bigtreetech TMC 2130 driver (SPI mode)  .

  I do not know what the main reason was !

But... the most important note is: when you enabled in the configuration.h X/Y/Z/E0  as Driver, you must connect the ((four motors driver, if you enabled E1 too you must connect the Fifth Driver  too  )) exclusively. If you do not , you will still have message ((TMC connection Error)).

My printer mode is:Tevo Tornado , SKR1.3 ,Bigtreetech TMC 2130 driver (SPI mode), Bltouch, and sensorless homing for X and Y Axis , displays are reprap discount and Bigtreetech TFT35 V1.2

 important note for reprap discount display((The connectors must be inverted connected))

and here is best seller you will find: Not the only one, but the better  .
quick communication and fast delivery.
you will be always satisfied.
https://www.aliexpress.com/store/228623?spm=a2g0s.9042311.0.0.316c4c4dgA2VgP

Thank you all for your time.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2019-07-23 by Salim169

MikRoscope, 2019-08-19:
sorry had to open this! I get the same TMC connection error and I could not solve the issue. Here is M122 output: 

> Connecting...
Printer is now online.
SENDING:M122
X	Y
Enabled	false	false
Set current	800	800
RMS current	795	795
MAX current	1121	1121
Run current	25/31	25/31
Hold current	12/31	12/31
CS actual	31/31	31/31
PWM scale	255	255
vsense	1=.18	1=.18
stealthChop	true	true
msteps	0	0
tstep	4294967295	4294967295
pwm
threshold	0	0
[mm/s]	-	-
OT prewarn	true	true
OT prewarn has
been triggered	false	false
off time	15	15
blank time	54	54
hysteresis
-end	12	12
-start	8	8
Stallguard thrs	8	8
DRVSTATUS	X	Y
stallguard	X	X
sg_result	1023	1023
fsactive	X	X
stst	X	X
olb	X	X
ola	X	X
s2gb	X	X
s2ga	X	X
otpw	X	X
ot	X	X
Driver registers:
X	0xFF:FF:FF:FF	Bad response!
Y	0xFF:FF:FF:FF	Bad response!
Testing X connection... Error: All HIGH
Testing Y connection... Error: All HIGH

The power jumper is set to INT +5V, power supply is 12V, and I followed the steps in Bigtreetech video for sensorless homing with TMC 2031 SPI. The board and drivers were purchased from Bigtreetech store on aliexpress: https://www.aliexpress.com/item/32980090169.html?spm=a2g0s.9042311.0.0.47424c4dVavtHK

I soldered two pin headers to XDiag and YDiag on the driver, is that correct based on https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/Wiring%20diagram/SKR-V1.3-TMC2130-SPI-Mode.png?

[image]
[image]
[image]

MikRoscope, 2019-08-19:
> SW SPI enabled? Is the TMC driver itself set to SPI physically?

How to set the driver physically to SPI? I don't see any unsoldered jumper on my driver! :-?

Talha909, 2019-12-11:
> After several experiments onSKR1.3 board ,with Original TMC 2130 (SPI mode) from WATTEROTT, my problem was : i had always message on the display ((TMC connection Error)), and ended the first [...]

can you please show the configuration file of your printer.

3DSmitty, 2020-10-02:
Have you all read this documentation? You have do some soldering to enable SPI:

https://github.com/bigtreetech/BIGTREETECH-TMC2130-V3.0/blob/master/TMC2130-V3.0RM.pdf

jessepietila, 2023-05-19:
> I had to remove CLK Pin on the tmc5160 v1.1 & #define MINIMUM_STEPPER_PULSE 1 I was able to move Motors after this.

How to know ifd I need to do the same ?