bigtreetech/BIGTREETECH-SKR-V1.3 #49: Reset button dosent reset board

TimboNZ76, 2019-07-11:
Reset button does not reset my board and thermistor pins don't detect temperatures just read 0.0.

crencirs, 2019-07-11:
Hi,

I have the same problem with the reset switch. (Thermistor is okay.)
After testing for a few days, finally I found a way.

Wait at least 2 minutes after power is completely removed from the board.
Then it works by connecting the board again. Firmware downloads will also be available.

But the reset switch still does not work properly.
Eventually I give up the reset switch and turn the whole power off and back on. I've been waiting for two minutes.
**Let me know if anyone knows how to fix the reset switch!**

On a side note, it take about 2 minutes to connect the USB cable but with the external power less than two minutes. (Of course, the reset switch still does not work if I connect external power.)

One more thing, since the firmware is stored in the NAND flash, the board operates even if there is no SD card.

I wish you all the best. :)

TimboNZ76, 2019-07-15:
I plan to go over my old board with a microscope camera and see if I can find any broken solder connections etc to see what was happening as my board had a fault causing it to go into a reset loop.  There is a short somewhere on the board and I am going to hunt it down and fix it if I can.

cinealfa, 2019-07-24:
> I plan to go over my old board with a microscope camera and see if I can find any broken solder connections etc to see what was happening as my board had a fault causing it to go into a reset loop. There is a short somewhere on the board and I am going to hunt it down and fix it if I can.

It think it would be an open not a short...

dgrimal, 2019-07-24:
Having the exact same issues as OP. Voltage on the thermistor pairs is 3.3 with nothing connected and drops to 3.14 with known-good thermistors plugged in.