bigtreetech/BIGTREETECH-SKR-V1.3 #50: USB device not recognized

banjo200579, 2019-07-11:
Had some intermittent issue with uploading firmware from sd card. Now the board does not read anything from card and over usb is not recognized.
Help please

crencirs, 2019-07-11:
Hi,

I had a similar problem. Wait 2 minutes while power is completely removed and try to connect again?

Apparently it was good at first, but suddenly it was not recognized as a USB device. And the reset switch did not work too.

Of course I still can not, I still need two minutes..

banjo200579, 2019-07-11:
Tried that with no success

crencirs, 2019-07-11:
Do you use Marlin firmware?
Is the comment in the following two lines removed in the configuration.h?

#define SERIAL_PORT -1
#define SERIAL_PORT_2 0

banjo200579, 2019-07-11:
Followed the youtube video by Teaching Tech and set up marlin correctly. When it was kind of working I had a green led briefly turned on next to one of the small mosfet but not anymore.

crencirs, 2019-07-12:
As I know, the green light turns on when you use a fan, heater, or bed.
Anyway, I hope you solve the problem. I suffered from power problems, and I still have not solved them.

tvixen, 2019-07-13:
Are you using windows ?
If Yes, which version

banjo200579, 2019-07-13:
I have tried it on win 7 an win 10 with a bunch of SD cards. It will not update from sd card no matter what as the firmware.bin file remain there intact. As for now there is just red led on and that is it.

skr-000, 2019-07-13:
Hey there!
I have same issue on my Win7 - 64bit.
I noticed something my windows recognize the board only when it is an original firmware after I upload a new one from Atom the board dissapears.
Were you able to solve it?
I'm trying different approaches - no luck so far

skr-000, 2019-07-15:
https://github.com/MarlinFirmware/Marlin/issues/14006

cloudpol7, 2019-07-18:
I have the same problem with "USB device not recognized". It's probably faulty NUP2114 on board. In few days I will find replacement (hard to get same one for me) and try to fix this.

ripkrizbi, 2019-07-22:
I have exactly the same problem, flashed the new (and working) SKR 1.3 with a custom Marlin (3x2208 and 1xLV8729). The board refused to wake up. Then I reverted back to the stock firmware (copied the original FIRMWARE.CUR as firmware.bin) and the board remains in the state where it is not recognized by windows (USB device not recognized). 

Has anyone had any success in getting the board back to life? I have another one but I'm reluctant to give that one the same treatment...

banjo200579, 2019-07-22:
Hi guys just a small update. The seller is really stubborn to admit there are issues with this board. A red led on means for him a perfectly working unit despite the usb device error and inability to update firmware. Case opened on Aliexpress resulted in rejected claim for faulty item. A video of faulty board in action is required / such a stupid request / despite a picture was uploaded. I''m really fed up with this approach from the seller,  15£ out of pocket and a nice paper weight. A big warning for people out there. SKR 1.3, it is a big NO until they fix this issue.

arbitur, 2019-07-22:
I had this issue as well. I fixed it by formatting the SD card that came with the board then install the firmware.bin file to it directly through a SD card writer instead of through the skr-card. Then I inserted the SD card into the skr card and boot and it shows up in windows now and works.

ripkrizbi, 2019-07-22:
I also managed to solve my issue by recompiling Marlin and downloading the firmware using the SD card reader. After inserting the SD card with firmware.bin on it, I let the SKR sit for a while and it came up.

cloudpol7, 2019-07-23:
Problem solved!...at least in my case.
I had break circuit from R33 to Pin30 of LPC1768 as in SKR-V1.3-SCH.pdf.
It can be faulty PCB batch/ poor quality via in this area.
[image]
[SKR-V1.3-SCH.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/3423854/SKR-V1.3-SCH.pdf)

crencirs, 2019-07-24:
> Problem solved!...at least in my case.
> I had break circuit from R33 to Pin30 of LPC1768 as in SKR-V1.3-SCH.pdf.
> It can be faulty PCB batch/ poor quality via in this area.
> [image]
> [SKR-V1.3-SCH.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/3423854/SKR-V1.3-SCH.pdf)

Hi, it's very interesting. In my case, the reset switch doesn't work well. It seems to be cut off the power for a very short time but after that my board doesn't work.
So I have to detach the power cable (or usb cable) completely and wait about 2 minutes.
Then, it works well. Is there anything you can advise me about?

Thanks

cloudpol7, 2019-07-24:
> Hi, it's very interesting. In my case, the reset switch doesn't work well. It seems to be cut off the power for a very short time but after that my board doesn't work.
> So I have to detach the power [...]

First check is there connection as I wrote. I had same symptoms with USB like you in the beginning and after while it completely stops working. Reset could be another circuit break issue. You need to check is R20 and switch has connection to Pin17 on LPC and other side of R20 has connection to 3,3V.
[image]
[image]

crencirs, 2019-07-24:
> First check is there connection as I wrote. I had same symptoms with USB like you in the beginning and after while it completely stops working. Reset could be another circuit break issue. You need [...]

I checked PIN17 to R20. Connetion is OK and 3,17V. Even I didn't know what problem is, thank you for your help. :)

cloudpol7, 2019-07-27:
Some update:
After USB problem fix, after while my board completely stops booting up... I replaced LPC, flash bootloader and firmware but now after about 1h of printing LCD and neopixel led start failing.
So many problem with this board... I assume that PCB is so low quality that after heating up random routs on PCB stops conducting. Future repairs is pointless.

ghpicard, 2019-07-29:
In my case the problem was solved by simply using another SD card instead of the one that came with the board. I assumed that as 128MB SD cards have not been manufactured for quite a while, the ones that come with the board are the cheapest they can get and thus, not trustworthy at all.
I used an (old) 2GB SD card with tons of other data in, copied the firmware.bin in there, inserted the card and plugged the board in the USB port of my laptop. The board came up to life immediately.
So in my case I can say that the problem was due to a bad SD card.

ghpicard, 2019-07-29:
#54

mrkrynmdsco, 2019-08-03:
same issue here. I suspect that the data line TVS (D3) is broken. I wonder if we can still use the board without going thru the USB

crencirs, 2019-08-03:
I just bought one more board. From Amazon ...

The problem is that it is difficult to verify that the board is working properly. Whether or not usb works.
We do not know if usb is broken or if something else is broken.
At least we need LCD or TFT to see if the board is working.

And since the firmware has already been downloaded to the flash, the SD card does not need to be on the board. The sd card is only needed when downloading new firmware.
If the LCD connected to the board is displayed normally, then we can think of usb as a problem.

But can it be solved? That's why I bought one more board. Why did I buy the same thing again? Other 32bit boards are too expensive..

mrkrynmdsco, 2019-08-03:
isolating the TVS problem, same issue.. now it seems that the USB device of the LPC1798 is now broken..
D+: 3.0 V
D-: 0.2 V

not sure if this is what it should be.

mrkrynmdsco, 2019-08-03:
if this issue could not be resolved, I think the board is still usable.
Last time I checked the bootloader is still working (tried to see if a new firmware in the SD card will be renamed).
USB connectivity is gone.

mrkrynmdsco, 2019-08-03:
This what worked for my case:
* Removed the TVS (D3)
* Revert the original firmware (re-built from the provided SKR 1.3 Marlin 2.0 repo)

D3 removal means vulnerable to voltage transient spikes (during hot-plugging).

MikeF59, 2019-08-06:
> In my case the problem was solved by simply using another SD card instead of the one that came with the board. I assumed that as 128MB SD cards have not been manufactured for quite a while, the ones [...]

Same for me. I used an 8 gb new one  and all is ok. Another problem is the reset button. After I measure it with an ohmetter I saw that is broken.  I made a bridge between two legs of reset button and the board resetted... hope help.

cloudpol7, 2019-08-14:
I found another problem and another solution.
After "USB unrecognized device" problem fix i get random characters and artifacts on LCD, Neopixel LED control problem, TMC comunication error and again USB problem.
The problem was to high resistance in routs from 3,3V to VDD pind on LPC1768. 
I measured voltage on C4 and it was below 2,6V and dropping after board heat up.
After I made another external route like in the picture (additional wire for 3,3V to LPC VDD) all problems was fixed. Even USB is recognized every time now. Time will tell how long it will work until another problem shows up :)
[image]

KrX3D, 2019-08-22:
@cloudpol7 
hi, i got the same usb unknown error after i replaced the lpc1768. i think i need to burn the bootloader like you did but i dont know how. could you tell me how you did it? what hardware/software do i need

cloudpol7, 2019-08-22:
> @cloudpol7
> hi, i got the same usb unknown error after i replaced the lpc1768. i think i need to burn the bootloader like you did but i dont know how. could you tell me how you did it? what hardware/software do i need

Hi, 
It's instruction with needed files:
[instruction ](http://smoothieware.org/flashing-the-bootloader)

In shortcut:
- Serial-USB ( can be a FTDI cable, an arduino, CP2102, etc ) cable connected to the UART0 pins of the LPC1769, P7- pin2,3 
[image]

- Program: FlashMagic
- bootloader file: [bootloader](https://github.com/Smoothieware/Smoothieware/tree/edge/bootloader)

KrX3D, 2019-08-22:
> Hi,
> It's instruction with needed files:
> [instruction ](http://smoothieware.org/flashing-the-bootloader)
>
> In shortcut:
>
>     * Serial-USB ( can be a FTDI cable, an arduino, CP2102, etc ) cable [...]

thank you very much. i tried this instruction yesterday but didnt see P7 so i couldnt do anythink. so thanks again for pointing me int the right direction.

after flashing the bootloader the usb device is recognized and marlin works again ;)

ghpicard, 2019-08-22:
It is interesting, but I also identified that enabling the SD card functionality in Marlin 2.0 (#define SDSUPPORT in config.h) together with #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is something that renders the board unusable, but not bricked. USB will not be recognized, no image in the LCD screen, no sound, no response from the rotary encoder, etc. The SD card filesystem used to transfer the program is damaged too and it requires repair or better yet, a reformat. 
The way to revert to normal was for me to create a new firmware.bin with the SD card functionality disabled in config.h, and use it with the normal firmware programming process.

Zviki68, 2019-08-28:
@Claudpol7 You used multicore wire for soldering to the processor pins?! Heh...

cloudpol7, 2019-08-29:
> @cloudpol7 You used multicore wire for soldering to the processor pins?! Heh...

No, I used kynar wire. Wire in picture was just for short therm test. Don't be fussy for trying to help others.

pmozbert, 2019-08-31:
replacing the SD card fixed this for me.  The stock card is known to be junk.

erencoo, 2019-10-17:
ı have a same issue I replaced sd card ıt didnt work

ultranoobian, 2019-10-21:
I was having the problem of putting new `firmware.bin` onto the supplied 128 MB SD card, then the board would not be detected in Windows 10.

I fixed it by removing the supplied microSD card and putting a new trusted microsd card with the `firmware.bin` file. Once the board powered up, It popped up again in Windows with my new sd card and my firmware.bin file replaced with firmware.cur

anggiatm, 2019-11-16:
hello i have same problem, i found the thermister of hotend is broken (short circuit), problem solved after replace the thermister. i suggest to unplug all attachment like thermister, heater, driver, etc before connect to pc. broken attachment can distrubing the controller. i think

mohacs, 2019-12-28:
> In my case the problem was solved by simply using another SD card instead of the one that came with the board. I assumed that as 128MB SD cards have not been manufactured for quite a while, the ones [...]

Same for me! Thank you.

kasimsuzen, 2020-01-16:
> I was having the problem of putting new `firmware.bin` onto the supplied 128 MB SD card, then the board would not be detected in Windows 10.
> 
> I fixed it by removing the supplied microSD card and putting a new trusted microsd card with the `firmware.bin` file. Once the board powered up, It popped up again in Windows with my new sd card and my firmware.bin file was replaced with firmware.cur

For me the cable was faulty. Took 6 hours of trial and failure before i try another cable

gadgetman48, 2020-03-20:
#define SERIAL_PORT -1 
 this one worked for me so i can see sd card through usb to pc.

leanlibutti, 2020-04-16:
> 
> 
> hello i have same problem, i found the thermister of hotend is broken (short circuit), problem solved after replace the thermister. i suggest to unplug all attachment like thermister, heater, driver, etc before connect to pc. broken attachment can distrubing the controller. i think

Hello! 
what is the thermistor of the hotend?
Regards

PiDocJohnson, 2020-05-31:
Just a side note,; for me, my three SRK boards are all very picky about the USB cable, if I use a cheap cable, no USB com port and no disk drive; swap out the cable with high quality cable; works perfectly. May not be your issue, but it's a worth a quick try. Belkin's seem to work all the time.

bendit22, 2020-08-17:
> Just a side note,; for me, my three SRK boards are all very picky about the USB cable, if I use a cheap cable, no USB com port and no disk drive; swap out the cable with high quality cable; works perfectly. May not be your issue, but it's a worth a quick try. Belkin's seem to work all the time.

OMFG, Its always the simplest thing. after having intermittent connection issues with windows recognizing the skr1.3 board then  finally not working at all over USB. replaced with all SD cards within a 5km radius, trying ten different marlin configs, watching all the usb diagnostic videos on youtube only to read this last message of maybe its the cable, replace the cable with the shitty blue one you get with the board and Bob's your uncle. showed up straight away. i am also still experiencing an issue with the reset button but that can be be worked around. i was using a 5m usb extention cable  and i guess there was a voltage drop over that distance. cheers PiDocJohnson, you saved me launching the printer out the window but there goes 6 hours of my life i will never get back.

tarifari, 2020-11-26:
> This what worked for my case:
> 
>     * Removed the TVS (D3)
> 
>     * Revert the original firmware (re-built from the provided SKR 1.3 Marlin 2.0 repo)
> 
> 
> D3 removal means vulnerable to voltage transient spikes (during hot-plugging).

Hello, i have the same problem!

Think i burned my D3. USB is not recognizing my board correctly anymore. firmware update via sd-card is possible

Did you just removed the D3 and the board worked again? Is USB- connection possible after removing it?

xxDOOTxx, 2020-12-22:
Hi there! a similar problem here too - no USB connection; finaly I found the error - the USB data lines D + and D- were short-circuited; I fixed it and now it works! 
[image]
[image]

mrkrynmdsco, 2021-01-22:
> Hello, i have the same problem!
>
> Think i burned my D3. USB is not recognizing my board correctly anymore. firmware update via sd-card is possible
>
> Did you just removed the D3 and the board worked [...]

My apologies for late reply (too late maybe). My experience was after removing D3 the USB connection works fine and everything is OK. I have been able to play around with the board (design/build my own printer). Until I unplugged from the USB port and plugged-in again, then my board is fried. I can't get everything working. Maybe the transient kicked in and D3 is nowhere to protect the board. I assumed the microcontroller is also damaged.

I suspect that most SKR 1.3/1.4 boards use these type of components for the USB section. It lacks stability and robust protection.

dalton908, 2021-07-07:
Hi, I have this problem too. Did you find a solution to this problem?I think this problem can be solved by re-uploading the bootloader. Help me to upload the bootloader to the SKR board.

Balckandy, 2022-10-19:
> It is interesting, but I also identified that enabling the SD card functionality in Marlin 2.0 (#define SDSUPPORT in config.h) together with #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is [...]

i have this exactly same problem,  afther enables those option the board totally brick i thought thas wass the bootloader but when i try to re upload whit st link didint recognice it but wen i pluged the sd its transform the . bin to .cur .  i hasen´t  idea what was going on but this comment really help me a lot thank u so  much