bigtreetech/BIGTREETECH-SKR-V1.3 #51: if enable LASER_FEATURE

sn0wrider, 2019-07-12:
Hi if if enable LASER_FEATURE. Where is PIN PWM 5v?

.platformio\packages\framework-arduinoststm32\cores\arduino/pins_arduino.h:223:51: error: missing 
binary operator before token "("
 #define digitalPinHasPWM(p)         (pin_in_pinmap(digitalPinToPinName(p), PinMap_PWM))
                                                   ^
Marlin\src\HAL\HAL_STM32\../../inc/../HAL/./HAL_STM32/fastio_STM32.h:83:33: note: in expansion of macro 'digitalPinHasPWM'
 #define PWM_PIN(P)              digitalPinHasPWM(P)
                                 ^~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:2332:12: note: in expansion of macro 'PWM_PIN'
     #elif !PWM_PIN(SPINDLE_LASER_PWM_PIN)
            ^~~~~~~
In file included from C:\users\administrator.home\.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:48:0,
                 from Marlin\src\HAL\HAL_STM32\../../inc/../HAL/./HAL_STM32/../shared/Marduino.h:34,
                 from Marlin\src\HAL\HAL_STM32\../../inc/../HAL/./HAL_STM32/HAL.h:28,
                 from Marlin\src\HAL\HAL_STM32\../../inc/../HAL/HAL.h:52,
                 from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:30,
                 from Marlin\src\HAL\HAL_STM32\fastio_STM32.cpp:25:

sn0wrider, 2019-07-12:
Im try   #define SPINDLE_LASER_PWM_PIN     SERVO0_PIN 
not working !!!!

sn0wrider, 2019-07-12:
pls close its for PRO v1.1

xoniq, 2019-08-01:
You can close your own issues yourself.

closed 2019-08-01 by sn0wrider