bigtreetech/BIGTREETECH-SKR-V1.3 #54: Stock SD card, totally junk

tvixen, 2019-07-13:
First time boot up went ok. Then tried to put new firmware on the SD card, and it went in to a totally waiste of time.

Just had a hard time to transfeer the firmware to the SKR 1.3. (5 hours)
Did try many different tips and tricks. It now appears the the stock SD card on 128Mb, is pure junk.
If you set it in a cardreader, and transfeer a file to it, the transfeer speed is under 1kbs.  This will not be accepted by the SKR 1.3 SD card reader, and the hardware hang. 

I now have fitted in a new 16Gb HC1 SDcard into the board, and everything works now.
So please...check your SD cards from vendor.... it's junk.

skr-000, 2019-07-13:
Unfortunately - I agree

carl1961, 2019-07-14:
so far I have no issues, have two boards a skr 1.3 and a skr Pro 1.1   tested it with H2testw
Test finished without errors.
You can now delete the test files *.h2w or verify them again.
Writing speed: 4.02 MByte/s
Reading speed: 15.6 MByte/s
H2testw v1.4
[image]

tvixen, 2019-07-15:
I guess you was lucky.

xoniq, 2019-08-16:
This is always with cheap cards which are mostly meant to be used to just setup the firmware. (So that you have at least a card to set stuff up)

I had the same thing with my previous 3D printer, thankfully a local store sells 32GB class 10 cards for 9 euro's so I bought one there, and no problems at all.

rkapl, 2019-11-24:
I second that. I was worried that I've ended up with a bricked SKR, while the stock card was just not able to hold the bits. Some sectors were corrupt after copying the firmware and re-mounting the card.