bigtreetech/BIGTREETECH-SKR-V1.3 #55: TFT 3.5 won't connect to board, it says no printer attached.

lowkin18, 2019-07-14:
I uploaded the new firmware onto the board and the firmware onto the TFT3.5 installed the com-port.

The board shows up as a USB drive as well, which is probably the issue and it's blocking the TFT3.5 and I don't know how to remove this.

mpallen509, 2019-07-16:
When the board flashes the SKR v1.3 firmware it renames the firmware.bin to firmware.cur. This makes sure that it doesn't rewrite the next power cycle.

Uploading firmware to the TFT3.5 should be the same manner. Using a SD card and a file titled: 

BIQU_TFT35_V1.2.1.0.bin

This does not get renamed after flashing. So remove it from the SD card after flashing so each boot does not result in more wait time. 

Also when connected to 12v power and plugged in via USB to a computer. The onboard microSD card will be displayed as storage. Mine displays the name of the microSD and allows me to copy/delete files from it.

lowkin18, 2019-07-16:
Mine displays as a USB marlin sd card and I cannot write to it. It wants me
to format it. But the only thing I care about is that my TFT 3.5 says no
printer connected when its clearly plugged into the board. I have serial 1
as -1 and serial 2 as 0. But I cannot get the tft to connect to the
board/printer.

lowkin18, 2019-07-16:
It did initially connect but once I installed the USB device from the
smoothiware file it started not finding it. But then I could connect to my
computer through pronterface and the com port

mpallen509, 2019-07-16:
How do you have the TFT35 plugged into the control board?

I utilize the TFT stand-alone pins just above the generic serial pins.

lowkin18, 2019-07-17:
I am using these pins as well. I have no other display connected.

Is there a way to hard reset the boards, or restore them to factory
settings?

lowkin18, 2019-07-17:
Does, there need to be any software modifications in the marlin firmware
filesfor the skr v1.3 for the TFT?

lowkin18, 2019-07-17:
I got the screen working with my SKR v1.1 pro which let me narrow it down
to the SKR V1.3 being at fault, so I checked the SD card and reformatted
it, then changed the SD card and now it works. so the original SD card was
corrupted and it keeps getting corrupted.

mpallen509, 2019-07-17:
Sounds like the card is a poor data type or is getting constantly reloaded until Windows throws a fault that the card is corrupt. 

Glad to hear it is working with a different SD.

tvixen, 2019-07-23:
Same as a lot of other users #54

bigrednurse, 2019-07-24:
I just bought one of these boards with the tft35 display.  I have been reading up before getting too  far, but I read that you can not actually use the card slot on the display d/t the address being in conflict with addressing on the skr board.

xoniq, 2019-08-16:
> I just bought one of these boards with the tft35 display. I have been reading up before getting too far, but I read that you can not actually use the card slot on the display d/t the address being in conflict with addressing on the skr board.

I am actually able to use the TFT SD slot, working just fine. I also have a smart LCD with knob attached, no problem. I only had a problem with the TFT wire not connect properly sometimes, failing the print (lack of stable connection), now i'm using the SD slot from the SKR itself to print, seems to be the most solid solution.

emilhse, 2019-09-25:
Definitely it's a firmware mistake, it initiates error with out checking 

BIGTREETECH-TouchScreenFirmware-master\TFT\src\User\Menu\menu.c
line 63: static REMINDER reminder = {{0, 0, LCD_WIDTH, TITLE_END_Y}, 0, STATUS_UNCONNECT, LABEL_UNCONNECTED};  there is no if or else statement so it will be on the screen forever.

BIGTREETECH-TouchScreenFirmware-master\TFT\src\User\API\Language\language_en.h
 Line 72:   #define EN_UNCONNECTED          "No printer attached!"

dineshvyas, 2019-10-04:
Take a new SD Card copy firmware on new sd card and flash firmware.  TFT35 should light up now. You may see a prompt of hardcoded temperature or something else. I made it work.

danielmar2160, 2019-10-06:
mine display all the time says No printer attached ! please help. TFT35 V3.0 i cant get ti worked :(

dineshvyas, 2019-10-07:
> mine display all the time says No printer attached ! please help. TFT35 V3.0 i cant get ti worked :(
[image]
this worked for me.
take a new sd card or format sd card.

danielmar2160, 2019-10-08:
This software is for skr pro but i have skr 1.3

Daniel Marczak
DMcircuit
www.dmcircuit.co.uk

pdschroder, 2020-03-03:
Should the TFT35 V2.0 work as it comes out of the box with both Marlin and Smoothieware?  My SKR V1.3 apparently came with Marlin, and the TFT35 connected to the board.  But as soon as I loaded Smoothieware from the BIQU site, I got the "No printer attached" error.  I haven't been able to get the TFT35 to see the board with Marlin either.

damenzijlstra, 2020-03-29:
I had a simular problem and had to bring back the connection serial speed from 250000 to 115200 in settings on the tft menu

billiamstearns, 2020-04-10:
Well My TFT24 also tells me "No Printer Attached"....and I have tried EVERYTHING YOU ALL SUGGEST....can I install my old Anet LCD12864

damenzijlstra, 2020-04-10:
hi i rewired the 10 pin ribbon cable to enable the anet A6 full graphics lcd to a BTT SKR mini E3. If you follow a similar process you should be able to do it also with the SKR V1.3. her is the link to my reddit post explaining what i dit
 Good luck with it. 

https://www.reddit.com/r/BIGTREETECH/comments/ftuoc3/anet_a6_full_graphics_lcd_on_skr_mini_e3_v12/?utm_source=share&utm_medium=ios_app&utm_name=iossmf

Verstuurd vanaf mijn iPhone

tohendo, 2020-04-21:
I had this problem for ages (and in a way still do). When firing up the printer the TFT would present "no printer attached" but it would go away once I had plugged the usb into my laptop. It would then work even with the usb removed. Still looking for a work around to get it to work without the need to plug it into the laptop on start up.

Newtonn2, 2020-04-21:
I'm having the same issue with my SKR V1.3 and the TFT24. I also tried everything I found here and on other webpages with no success. I decided to connect the oscilloscope to the TX pin of the TFT24 and it seems to be working sending the M105 gcode every few seconds. Connecting the oscilloscope to the RD doesn't show any activity at all... so I'm pretty sure its to do with the firmware as it seems that the board its not responding to the screen. I have tried installing Marlin from scratch and doing all the necessary adjustments... changing SERIAL_PORT number, etc. I have TMC2209 and BLTouch. everything works apart from the LCD. I hope someone can share some light on this issue as I don't know what else to try to get the touch screen to work.

pdschroder, 2020-04-21:
> Caution! This email is from outside our organization.

Sorry--I can't help.  Mine started working after I updated the firmware on the display.  The plug to the SKR 1.3 is kind of ambiguous--I assume you have double checked that.

phiphe, 2020-05-08:
I
Hi,

Last week i bought a bigtreetech TFT 24 V1.1 and skr 1.3 board.
I put marlin 2.0.X: 
#define SERIAL_PORT -1 
#define SERIAL_PORT_2 0 
#define BAUDRATE 115200

#define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3

I change jumper in skr to USB power.
I connect 1pc touch screen cable and 2 pcs ordinary LCD cables.
I run all with usb cable….update tft24 menú. 

The classic marlin screen works fine.....if i connect with simplify 3d, all works ok....

The problem is when i put touch mode …...always says "no printer attached"....i put 115200 in menú too….but no communication...

If i quit the 5 pin cable don´t happens nothing…..is how no communicate with skr 1.3.....i try with 5 bltouch cables and doesn,t work too....
nothing resolve it?

RavenWarrior, 2020-05-30:
I’m getting the same message, no printer attached, I’ve tried everything you guys have and it won’t work. Just adding more fuel to the fire here, in hopes of a resolution

dirgesan, 2020-06-06:
I just received my BTT Turbo 1.4 and TFT35 Thursday and updated both on Friday. The hardest part was learning Platform IO. To have the TFT35 communicate with the board you need to add a serial port for it in the printer controller board. My Turbo 1.4 only had 2 by default, one for the bootloader and another for the USB. With the USB plugged in the TFT35 can't talk to the controller board. Just had #define SERIAL_PORT_3 1 and recompile the firmware. I have -1,0, and 1 as my bootloader, USB, and TFT35 ports, respectively.

RavenWarrior, 2020-06-06:
Turns out my tft35 is knackered... the tx, rx and gnd are shorting on one another and with the power and reset pins...
Biqu are sending another.
So if your screen isn’t working, I’m assuming the pins shouldn’t short. Maybe only power and gnd, and little blip from reset and gnd.
I don’t know what I’m talking about though.
Someone else would need to reiterate what’s it actually should be.

dirgesan, 2020-06-06:
Yeah, the pins shouldn't short to zero ohms. So I take it your board doesn't boot. What should happen is both the board and TFT boot. Then the TFT shows the "No printer attached!" for a few secs until they establish a connection. Glad to hear they're sending a new one.

RavenWarrior, 2020-06-06:
> Yeah, the pins shouldn't short to zero ohms. So I take it your board doesn't boot. What should happen is both the board and TFT boot. Then the TFT shows the "No printer attached!" for a few secs until they establish a connection. Glad to hear they're sending a new one.

So the board boots accepts firmware and so does the lcd.
The screen still displays. And still powers. But just doesn’t connect. ‘No printer attached’ continuously...
So if anyone is having this issue, and none of the baud rate changes and port changes are helping, or commenting back in something you’ve uncommented that interferes With the lcd, maybe check for continuity between rx tx and gnd.
Biqu will require proof, I.e video of you testing the board...
They blamed me of course but only by stating this next lcd is the last one. They apparently check them all before shipping so it’s my fault there’s a short.
Funny cos I only plugged it in and uploaded marlin.

dirgesan, 2020-06-06:
So you say you added a third serial port in the controller board firmware and it still doesn't work. You're probably right, a bad board. I haven't put mine in yet. Still building the printer for it. I noticed the black Dupont cable doesn't make a very good connection. I'll build a JST cable of the correct length once everything's built and use only the touchscreen mode. The aux cables are too short and too much of a PITA to build.

nilspear, 2020-06-24:
try this:
#define SERIAL_PORT 2
this works for me using a TFT35-E3 V3.0  and a SKR-MINI-E3 V2.0 board.

craigistired, 2020-06-27:
FYI: Settings for TFT35 V2.0 to work on the SKR Mini (with integrated TMC2209 drivers).
#define SERIAL_PORT 2 //Set to 2 to get screen to connect to printer
#define SERIAL_PORT_2 -1 //Set to -1 to get octoprint to connect to printer

Newtonn2, 2020-06-27:
This is not my issue as I have tried those settings without success. 😞

meltonpieman, 2020-06-27:
Like Newtonn2 I have tried everything suggested on this thread and all works well: BLTouch, TMC2209 and LCD mode of my TFT35 E3 v3 screen. Just it will still not recognise the printer. Today I have recompiled the BTT firmware to the latest mods (from their Github site) and installed the resulting BIN file - all fine. BUT still no touch screen connection to the printer!

octico, 2020-07-04:
> Like Newtonn2 I have tried everything suggested on this thread and all works well: BLTouch, TMC2209 and LCD mode of my TFT35 E3 v3 screen. Just it will still not recognise the printer. Today I have [...]

I had the same issue, did ou check your baud rate under UART? It should be set 112K

brbubba, 2020-08-04:
> 
> 
> FYI: Settings for TFT35 V2.0 to work on the SKR Mini (with integrated TMC2209 drivers).
> #define SERIAL_PORT 2 //Set to 2 to get screen to connect to printer
> #define SERIAL_PORT_2 -1 //Set to -1 to get octoprint to connect to printer

Thank you this worked for me after pulling my hair out! Needed port and now it finally connects to printer.

cmicheli346, 2020-08-28:
hi, i too have this big communication problem between TFT 3.5 V2 and SKR 1.4 turbo (drivers TMC2209).
I have tried everything. I can't set #define SERIAL_PORT 2 because it doesn't compile, "Serial port pins (2) conflict with E stepper pins!". With ports 1 and 3 instead it compiles but still does not connect. It didn't even help to declare #define SERIAL_PORT_3 0.
A strange thing, if I connect the printer to the pc and try to connect with Simplify3D or Pronterface it connects normally but the TFT keeps showing not connected, instead if I connect with Repetier-Host the TFT seems to connect, tft shows "ready" and also the temperatures but does not execute any commands and does not show coordinates or fan speed etc. Do you really have to struggle so hard to connect a display?

Ohmarinus, 2020-08-30:
Same here, my TFT 35 v3 screen is quite new and all of a sudden doesn't want to connect anymore. I haven't done anything to it. Earlier today it suddenly didn't want to connect anymore intermittently and now completely unable to connect. I guess there is something inside that is broken. I hope BigTreeTech is willing to send a replacement.

dirgesan, 2020-08-30:
I learned a few things setting up my TFT35 and BTT1.4T. Check all the connections. If something like a thermistor is configured in Marlin, and not connected, the board shuts down and the TFT won't connect. I had mine working for a few weeks and then it suddenly stopped connecting. The supplied MicroSD card had failed. I only found this out when I tried to use that same MicroSD to update the TFT35, and kept getting "file not found" errors when flashing the firmware. The supplied MicroSD is garbage. I don't know if the bad card inserted in the 1.4T was causing the connection loss or if I had re-flashed the 1.4T with bad firmware, but replacing the card with a good one solved my problem. Try re-flashing with a good card and see if that helps.

ZhuDaHai, 2020-09-10:
I have a the TFT35 v3.0 attached to an MKS SGEN L v2.0 board on Aux-1, with TMC2209s in UART and OctoPi  and these settings in configuration.h worked for me:

```
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
```

Hope this helps somone.

LegendArmour, 2020-10-09:
> FYI: Settings for TFT35 V2.0 to work on the SKR Mini (with integrated TMC2209 drivers).
> #define SERIAL_PORT 2 //Set to 2 to get screen to connect to printer
> #define SERIAL_PORT_2 -1 //Set to -1 to get octoprint to connect to printer

I tried all sorts and it was this that fixed it.. I was using serial_port 1 and switched between -1 when all I had to do was select 
 #define SERIAL_PORT 2

sozruh, 2021-02-14:
Have the same problem with skr pro v1.2 and TFT3.5 V3.0.

Anybody have this combination and a solution?

thanks in advance

cmicheli346, 2021-02-14:
changed display and everything worked at the first time .... so i think i had defective tft

harashtaw, 2021-03-12:
Happened to me earlier this week. TFT 3.5 suddenly won't connect to skr 1.4. Have tried to re flash the board firmware, format skr sd card, check and changing wiring, no luck. BUT it turns out that I've accidentally change the TFT UART baudrate directly from the screen itself when I manually changing machine setting for stepsize directly using the touchscreen. Maybe re checking the UART baudrate using the touchscreen and matching it may help  because that's the source of my problem.

BoomerT6, 2021-05-06:
I'm having the same problem with my BTT TFT 24. If an when it does connect, it states '3D Printer Ready', shows the bed and hotend temps etc, but it's a communication one-way-street, as any commands such as pre-heat or move, are not executed.

This problem seems all too common. Either Bigtreetech are failing to tell us something we need to know, or the BTT TFT 24 and BTT TFT 3.5 displays are seriously flawed and not fit for purpose!