bigtreetech/BIGTREETECH-SKR-V1.3 #56: MISO and MOSI shorted

Dendrowen, 2019-07-16:
Yesterday I blew up something on my board and now some functions are not working anymore like the PWM signals to the bed, hotend and fans. I ordered a new one already.

Today my Watterott TMC2130 arrived and I tried to make them work but... Too bad. I get a 'bad response' from the M122 command after trying several settings. 

While debugging I tested the board and saw that the MISO and MOSI signal (SDI and SDO on the TMC) are shorted. Could someone tell me if that is the result of the board that blew up or if this is normal? I'd like to know if I can continue testing the TMC2130.

tvixen, 2019-07-23:
These two should NOT be shorted. It's two different things /ports.  One is input and one is output.

Dendrowen, 2019-07-23:
That's what I thought. Thank you for clarifying.

closed 2019-07-23 by Dendrowen