bigtreetech/BIGTREETECH-SKR-V1.3 #58: Config.txt from MKS Sbase not working

Curedbio, 2019-07-20:
Using Config.txt from MKS Base v1.3 on SKR 1.3 (Win10). OS dont recognize board, only config.txt from Bigtreetrech.

bleughb, 2019-07-20:
MKS Sbase is a different board, different manufacturer, different construction.

https://www.instructables.com/id/Configuring-MKS-Sbase-V12-32-bit-Controller-Basics/

Curedbio, 2019-07-23:
but it uses smoothiware firmware... the same as the original smoothie and mks sbase, no?

tvixen, 2019-07-23:
I can't see the similarity, why do you think it would work?   It's 2 different boards.
I wouldn't raise this as an issue.

bleughb, 2019-07-23:
> thiware firmware... the same as the original

manually copy items?
you using smoothieware or marlin?

contact MKS

Filcanna, 2019-07-23:
Same for me. I had to rewrite/adapt the config file. Dunno if it's a length problem (SBase one was 29 KB, SKR 22 KB) or other. Maybe also that my Macs are adding something, to config files, unbearable to RKR board. Out of this Smoothie on SKR behaves apparently the same as on SBase (at least for menus, still not tried in production printers)

Curedbio, 2019-07-29:
Copy config file manuallly to SKR sdcard.
Using smoothiware, i like how easy is to config the board.
Can be a UTF format file problem?

Curedbio, 2019-07-30:
Setup uart0.baud_rate 115200 and disabled all network related. 
Now it connects the way it should. Stepper motors pins reconfigured.

closed 2019-07-30 by Curedbio

Filcanna, 2019-07-31:
A similar problem occurs with other Smoothie-cards (like Azteeg Mini) NOT having ethernet connectivity on PCB; they all seem to crash when SBase config.txt is used, if set to have network enabled. So simply putting false on network is able to solve. http://forum.smoothieware.org/forum/t-6085172/config-txt-causes-azteeg-x5-mini-v3-not-to-boot