bigtreetech/BIGTREETECH-SKR-V1.3 #59: TMV2209 support

Dilbert0815, 2019-07-26:
I had problems getting TMC2209 step sticks working with UART configuratio in Marlin 2.x..
Some Bigtree TMC2208 before were working correctly, so I was debugging the case.

Problem is on these TMC2209 sticks the port RST1 is wired directly to PDN_UART from TMC2209 while M3 port is having an addtional 1k ohm resistor to PDN_UART of TMC chip..
So using the normal UART jumper on SKR1.3 we get a total of 2k (1k + 1k) in the TMC TX line, causing it to fail.

Hardware Fix:
Fix is possible in hardware by using a adtional jumper in SPI jumper block between pins 1/3.
This shortens RSTI and M3 ports of step stick, effectively disabling the stick internal 1k ohm resistor. UART jumper is of cause also set.
Communication is now working and TMC_DEBUG with M122 returns correct register values and reports TMC2209 communication as online.

May be you can add this to board FAQ for troubleshooting TCM2209 stick problems of other users.

Dilbert0815, 2019-07-26:
Jumper illustration on X-stepper port:

[image]

NaterNore, 2019-08-07:
Hey, I was wondering if I could get these to work on the skr 1.3 non pro version. That mod you made is that little jumper I'm assuming? And this would work on any standard tcm2209? Thanks I'm a bit new to this.

ArchSnatcher, 2019-08-16:
Thanks for this advice Dilbert. I've been getting an "all Low" for Y on my 2208s and a "TMC connection error" on the LCD for the last week. I've been through countless pages of issues that are similar to mine, but we're not able to fix the problem.
I'm going to try this after work to see if it is an extra resistor. I'm using a total of 4, Eryone 2208 v1.2 drivers.

Dilbert0815, 2019-08-16:
> Hey, I was wondering if I could get these to work on the skr 1.3 non pro version. That mod you made is that little jumper I'm assuming? And this would work on any standard tcm2209? Thanks I'm a bit new to this.

Actually this IS a SKR1.3 non-pro. The Pro is actually revision v1.1.
 
But there ist no standard for TMC220x. Basically any vendor can route secondary signals to any unused port. This is exactly what happens here with Rx/Tx. 
Bigtreetech has routed their board for THEIR TMC steppsticks, and thats obviously not the idea that FySET had in mind when designing their 2209 sticks.
The TMC2209 is a chip standard, but the step stick is a vendor desgin and there are simply a few versions out there how to rout non-essetial ports to stick pins. This are ussually the stick pins that carry the setting pins on older chip designs.

You should have an ohmmeter ready and check how your sticks are routed to the TMC chip pins.

ArchSnatcher, 2019-08-17:
And it wasn't a resistor. The XYZ and E all work perfectly well when I turn on 'movement' mode on the LCD. I'm still getting the tmc connection error, but it is interesting. I have 4 total 2208s on xyze. When I run a check through terminal xy and z will be fine, but e will be 'All Low'. I'll move x to y (for example) and xze will be fine with a 'All Low' on  y.
I'm starting to think this is a tx/rx issue.

NaterNore, 2019-08-18:
So you're telling me that if I get the stepstick drivers I'd have to do this but if I get the bigtree or some other brand I might not?

jimmyeao, 2019-08-18:
Curious, do the 2209s still need the pads bridging like the 2208s did?

thisiskeithb, 2019-08-29:
To add to this, if you're using genuine Watterott 2209 V2s and would like to use sensorless homing without "flying wires", some hardware modification is required:
[image]

The Watterott 2209 V2's DIAG pin is in a different spot than what the SKR 1.3 is setup for on other/earlier drivers.

Disclaimer: This is just for general information and I'm not responsible for any damage to your driver/board/printer.

MarkV3082, 2019-09-04:
Dilbert0815 Thank you very much. This jumper config has solved my problem. Endless variations with these chips as you say.

asdmin, 2019-10-10:
@Dilbert0815: also, it seems that this mod with the cross-jumper drives the SPREAD at times when UART communication takes place. As I see, during UART the driver is 'required'  to rapidly switch between spreadCycle and stealthChop many times. Isn't it a problem?

Dilbert0815, 2019-10-18:
@asdmin : I don't have the board mounted acutally.
But there may be a misunderstanding. SPREAD ist not connected with this pin, at least not on the sticks I have here. If yous sticks have SPREAD connected to any of these pins do NOT nuse the mod.
Here on the FTYSET sticks have no other connection on these pins, they are both just UART.
Thes just put the RX/TX seperating resistor onto the stick and have the TX port on the wrong pin for the SKR board.

MS1/2 should have internal pull-down resistors according data sheet. It is normal to leave them open to my knowledge.

The-Monkey-King, 2019-11-08:
@Dilbert0815 : Please update this as a hardware fix for the FTYSET sticks. I am receiving the same error messages for the BTT SKR V1.3 and the BTT TMC2209s. I tried your pin fix and the TMC2209s from Big tree Tech still issue null responses from the X, Y, Z, and E0 registers. (0x00:00:00:00)

klubglied, 2019-12-15:
> I had problems getting TMC2209 step sticks working with UART configuratio in Marlin 2.x..
> Some Bigtree TMC2208 before were working correctly, so I was debugging the case.
>
> Problem is on these [...]

hi there.
i'm quite new to UART mode, linear advance and TMC's... therefore my BigTreeTech TMC2209's don't do what they should... with a M122 i get positive feedback from my printer (the connection is "OK" and all drivers send a sort of mac-address-number back) but when i try to print something (i tried a tube) the printer does not print the correct circle but an "interpolated" version of it (see in the picture below).
i tried to guess what is the problem here and changed some settings in marlin 2.0 (i am using a skr v1.3 by the way) for example the "MINIMUM_STEPPER_PULSE" to 2 because otherwise the E stepper would not move at all (seems to be a common problem with TMC's and linear advance) and the given work-around with the jumpers (your solution).
but still i don't fully grasp the problem...
can anybody help me please!?!

klubglied, 2019-12-15:
[image]

generalPanic, 2019-12-19:
I'm having issue w/BTT 2209s and my SKR 1.3 as well. Anybody have links to data sheets/shematics for BTT Stepper driver board?

klubglied, 2019-12-19:
hi generalPanic.
what exactly is your problem? I think I solved mine since my printer is doing what it should (although I did not test linear advance again).
what worked for me was to look up the specifications of the TMC's at AliExpress in the instructional video by "teaching tech" he says to set the"RSense" value to 0.075 but BTT's TMC's have 0.11 Ohm. maybe this caused the stuttering in my print. after I changed it and commented out the linear advance feature my ender 3 works perfectly again (tomorrow I will try to implement the LA-feature again! would be a shame to miss out on this...).

klubglied, 2019-12-19:
https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2

generalPanic, 2019-12-19:
Thanks for the link.  I have my 2209s inserted into my SKR 1.3 with the
UART jumper set, the SPI jumpers removed and I get all zeros when i do
gcode M122.

I changed the appropriate values in my Configuration.h (set drivers to
TMC2209 etc.).

Thanks,
Ben

klubglied, 2019-12-19:
hi Ben.
Just to make sure... did you uncomment the "#define x_driver_type TMC2209" (etc.) in "configuration. h"? what else did you change for your TMC's in Marlin?

generalPanic, 2019-12-19:
Yessir.

#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2209

BTW, I do not have the motor power or any motors  connected.  I'm using the
power from the USB port to power the stepper logic.

Cheers,
Ben

klubglied, 2019-12-19:
hi ben.

> BTW, I do not have the motor power or any motors  connected.  I'm using the
power from the USB port to power the stepper logic.

is there a reason you don't want to connect your DC input to the SKR?
i am not a great electronics wizard but maybe you should connect the real DC input (12V/24V) as each of your stepper drivers is trying to power the virtual stepper motor with about 0.8A-1.0A which would be too much for a normal USB power supply. but really i am just guessing here...

generalPanic, 2019-12-19:
I did connect 12v dc power, and this solved the problem. Thanks for your
help.

Cheers,
Ben

klubglied, 2019-12-20:
Hi Ben.
so great to hear that it works now! if you use your TMC's for a 3D printer I can strongly recommend to uncomment the "linear advance" feature in the "configuration_adv.h". if you do so you also have to change/uncomment the "#define MINIMUM_STEPPER_PULSE" and change it's value to 1 or 2 as the proposed "0" for TMC's let the E stepper motor quit working. (cf. https://github.com/MarlinFirmware/Marlin/issues/11024)
best.
Sebastian

hydra, 2024-07-14:
> I had problems getting TMC2209 step sticks working with UART configuratio in Marlin 2.x.. Some Bigtree TMC2208 before were working correctly, so I was debugging the case.

Many thanks for posting this!

I have a BTT SKR V1.3 with 5x BTT TMC2209 V1.3 drivers.

My symptoms were:
* very hot motors at idle - to hot to hold.
* M122 reported Error: All LOW for each stepper axis.
* High current draw, around ~3 amps on the bench PSU after issuing M17 to enable the steppers.

After removing the two jumpers on MS0 and MS1 and bridging the RSTI and M3 as above, I can report:
* the motors repond to GCode commands.
* current draw with all 5 stepper motors enabled is now only 382mA.
* the motors do not get hot.
* I get `Testing [X/Y/Z/I/J] connection... OK` for each axis.

I had two different motors:
* 4x Casun 42SHD0001-24B on X/Y/I/J.
* 1x Stepperonline 17HE15-1504S on Z.

I can't yet report if sensorless homing works, as there's nothing connected to the motor output spindles yet.