bigtreetech/BIGTREETECH-SKR-V1.3 #60: SKR 1.3 ender 3 slow heatign bed using 12v

PuiG18, 2019-07-30:
Hello, 
Does anybody know how to make the skr use 24v with the BED instead on 12v, i measured it while heating and doesnt pass 14v, (ender3 is 24v) i noticed when it took twice as long to heat it up.

Thank you

xoniq, 2019-08-01:
An option might be; using a cheap external mosfet board to put 24V to the bed. Then the SKR only puts 12v on the data (on/off) line towards the mosfet board.

Besides that workaround, the SKR1.3 should just pass 24V if you pass 24V to the board itself. Also check the 12V/24V output on the board itself if it handles the proper Voltage.

JimboMcRackem, 2019-08-15:
I've seen this. I had MKS GEN L and before that a generic board with A4988s and the bed was quick to heat, however with the SKR currently it takes 6+ minutes to heat to 50c from 25c and I'm using a MOSFET to power the bed. It seems that once the temperature gets to around a 4/5 ths of the value e.g. 40c with a target of 50c, the heater starts pulsing (reviewing LEDs) which slows heating dramatically. I can't find a Marlin setting for this. 

Anyone got any clues? This was with the latest firmware. I did choose the Tevo Tarantula Pro as a base configuration but will start from a clean config.h\config_adv.h file if it doesn't improve.

xoniq, 2019-08-16:
@JimboMcRackem 
> Scrub that - just found this which describes the issue.
> http://reprap.org/wiki/PID_Tuning
Lol, I read your message, and wanted to reply: "PID tuning", and then I read that you already found it.

PID tuning is the answer indeed. This lets the printer benchmark how much time it needs to reach a specific heating point, so it can anticipate on it. This has great benefits:

- **More efficient heating to the target**
  It knows when it reaches the target, based on room conditions, so it can heat faster and stop perfectly in time, instead of rush half and then slowly crawl upwards.

- **Better in maintaining heat at target**

  If you don't do PID tuning, you might see the heat bouncing between -5 and +5 degrees, which makes the firmware wait till it keeps steady for a few seconds. This can take a long time if the temperature keeps bouncing. With PID you keep a steady temp, which makes the print start sooner.

- **Energy efficient**

  Because it doesn't have to bounce temperatures, it just keeps it steady, which saves a bit of energy.

JimboMcRackem, 2019-08-16:
@xoniq 

Certainly makes a big difference, back to the 1 minute warm up time now.

Tomthi, 2019-11-21:
I'm having this same issue where the bed is super slow when using the PID settings. I've run the auto tune and put the values in but nothing is helping. When I switch to Bang Bang heating, the bed gets to 70c in  2.5 min. Which is great, but the temp swings are +- 3 degrees.  The PID settings are no where close. I can't figure out what I need to change to get better performance. The auto tune for the hotend worked very well. No issues with that.  Any help is appreciated. I read the linked articles but increasing the P and I  values didn't make any real difference.

JimboMcRackem, 2019-11-21:
I ended up using the PID values from an old configuration for my machine. Heating is just slightly slower than the hot end which is on a par with the original board. 

These are for the Tarantula Tevo settings taken from Jim Browns fork of Marlin. I couldn't use auto tune as it didn't settle in time with stock settings. 
 
  #define DEFAULT_bedKp 984.88
  #define DEFAULT_bedKi 193.91
  #define DEFAULT_bedKd 1250.55

closed 2019-11-21 by PuiG18