bigtreetech/BIGTREETECH-SKR-V1.3 #61: SKR 1.3 - Dead?

htzmedia, 2019-08-01:
Hi, after changing the Firmware 2 Times the Board is doing nothing anymore.

The Screen was already working, but now the Screen isn't working and the PC doesn't recognize the USB Device at all.

Is it possible to brick the Board? I don't know what i have done wrong.

htzmedia, 2019-08-01:
Formating SD-Card to Fat32 and rewrite firmware.bin solved the Problem.

xoniq, 2019-08-16:
> Formating SD-Card to Fat32 and rewrite firmware.bin solved the Problem.

If you are using the black micro SD which was delivered with the SKR board, toss it away, and buy a proper class 10 card. This makes sure you can rule it SD issues.

(Also, mark the issue as closed since it's solved, you can close your own issues)

closed 2019-08-16 by htzmedia