bigtreetech/BIGTREETECH-SKR-V1.3 #62: MKS TFT32 Issue

Reddemon159, 2019-08-03:
I have an MKS TFT32 v3.0.3 hooked up to the SKR v1.3. It powers up loads up and everything. But the display doesn't receive any info from the board. Such as Temps. Touch commands are unresponsive. I have it plugged into the AUX 1 Slot. Is it not compatible with this control board?

closed 2019-08-12 by Reddemon159

djtaotao, 2019-10-08:
How did you solve it?

djtaotao, 2019-10-09:
> How did you solve it?

Nevermind. I have solved.
It was a baudrate mismatch, between MKS TFT35  and Marlin.

edoluck90, 2021-03-09:
Please can you show a scheme to connect it ?  I have a mks tft32 v3 and a skr v1.4 turbo and I can’t connect because I don’t know how to wire it

djtaotao, 2021-03-09:
Connect the 8 wire ribbon cable to aux 1 on skr.

djtaotao, 2021-03-09:
I see only 8 pins on aux1.

edoluck90, 2021-03-09:
Ok but on skr v1.4 turbo you only have the tft port with 5 pin and the exp1 exp2 you have 10 pin

djtaotao, 2021-03-09:
Ok, I didn't know that skr turbo don't have aux1. But has a 5 pin connector and you can use 4 of them: RX, TX, Gnd and Pwr (5V).
[image]