bigtreetech/BIGTREETECH-SKR-V1.3 #64: SKR V1.3 + TMC2208-V3.0 + DRV8825 UART Bad Response

anjoze, 2019-08-09:
Hi,
I have a BIQU SKR V1.3
TMC2208-V3.0 for X and Y
DRV8825 to Z and E0
I have set the jumper to X-UART and Y-UART like in the instructions: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/Wiring%20diagram/SKR%20V1.3-TMC2208-UART-Mode.png
Then I've set the jumpers for Z and E0 like in the instructions: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/Wiring%20diagram/SKR%20V1.3-STEP-DIR-Mode.png

When I send M122 I always have X an Y enabled: false and "Bad response"
On TMC2208-V3 instruction page 5 says: "the user who purchases the UART mode driver does not need to perform any soldering operation"
( https://github.com/bigtreetech/BIGTREETECH-TMC2208-V3.0/blob/master/TMC2208-V3.0%20manual..pdf )

Do I have to solder or not? Thanks for any help.

FCosta1, 2019-08-11:
You have to solder the 2 pads...

anjoze, 2019-08-12:
Thank you for your answer.
Yes, I can confirm the **bad** instructions from bigtreetech.
On video, they just insert jumper on the board...
I've soldered J2 (right image) as described on 2.2 UART Working mode ( https://github.com/bigtreetech/BIGTREETECH-TMC2208-V3.0/blob/master/TMC2208-V3.0%20manual..pdf )

Dilbert0815, 2019-08-16:
Actually you may not need to solder.
Just follow my issue thread on TMC2209 support.
You can actually do the needed 'short circuit' extrernally in the jumper field below the step stick.
The problem here is only which of the two pins the TX signal is routed. By placing the external jumper you don't need to care which pin is selected by J1/2

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/59

anjoze, 2019-08-19:
Now I have it already soldered so I can't try.
Thanks any way Dilbert.

closed 2019-10-07 by anjoze