bigtreetech/BIGTREETECH-SKR-V1.3 #65: hardware problem?

FCosta1, 2019-08-11:
Hello..
I have a strange issue..
When i connect the skr 1.3 via USB the PC detects and the microSD appears and everything works ok...
When i connect 24v i loose comunication with the pc...
But..
If i press the reset button.. The USB works again...
The board only works when i press the reset button....
Hardware fault??

FCosta1, 2019-08-11:
I quit this board...
Put original firmware..
Connect the USB cable and 24v power supply..
Loose connection with the pc...
connect lcd2004 with original firmware... 
Appears:
TMC CONNECTION ERROR...?????
Without doing nothing.... 
Without doind any changes in marlin....
I quit this

korttwulfe, 2019-08-11:
There is a jumper near the middle of the board that says INT  +5v  USB.  When powering with your Power supply (not usb) make sure the jumper is on the INT side.

FCosta1, 2019-08-11:
I tried both ways and nothing...

xoniq, 2019-08-16:
Have you had the board up and running already once? Or is this a new experience? What stepper drivers are you using?

Also, when testing, do you have heat sensors attached? When I just got the board, I got crazy that the board stopped working after starting up, and it was because I had a test setup, but no temperature sensors attached so it shut down because of no temperature reading.

FCosta1, 2019-08-17:
The board only worked or by usb to access sdcard or by 24v to really work...
There is a component that is dead.. Waiting for it to swap to see if the problem is from that or is really dead the board... 
The component is:
MC74HCT125ADTR2G