bigtreetech/BIGTREETECH-SKR-V1.3 #71: Ender 3 with 2130's and BLTouch 3.0 - Z-axis doesnt go down, BLTouch doesnt work

JBrew13, 2019-08-22:
Looking for some guidance, I'm not the best with firmware, but I upgraded the board on the Ender 3 to the SKR v1.3, and got the TMC2130 stepper motor drivers. I went through the Teaching Tech videos, and quite a few others and going step by step with them, the printer didnt work to various degrees. I found a compilation for the Ender 3, which works the closest, but still not great. 

The issue's I'm having are that the Z-axis doesnt go down at all. When I go to Home, the BLTouch deploys a few times and I get the error on Pronterface to reset printer (M999). I dont quite understand what it is I'm doing wrong, I've swapped the 2130s so I know they are all good, the X and Y axis home, although very slowly, which I'm sure I can change in the firmware for homing_feedrate but just havent done it yet. But the Z is the big issue. If anyone knows whats going on with the BLTouch V3, great, otherwise I may just have to scrap it altogether. 

If anyone has advice, it'd be greatly appreciated. Like I said, firmware is not my forte, but I'm willing to learn what it is I need to look for, I just dont even know where to start here. Thanks in advance!

closed 2019-09-09 by JBrew13

cronos45, 2019-12-09:
Did you solve your issue @JBrew13 ?

nakedandjameson, 2020-03-16:
@JBrew13 Did you ever find a solution to this issue? I'm having the same problem.