bigtreetech/BIGTREETECH-SKR-V1.3 #73: missing info in pins for birtreetech skr v1.3

andracowolf, 2019-08-30:
I was messing with the 1.3 trying to get the CNC fan pin to power a fan and I noticed that the pin number was p2_3 but if you look in the pins file for the 1.3 if shows pin is simply fan_pin. 
Line 174 says 
#define FAN_PIN            P2_03 
when I tried to add that to config.h it gave me an error saying that you could not define a pin as fan_pin I went back to the pin file and on the line 174 I changed it to 
#define FAN2_PIN            P2_03 
 and in configuration.H changed line349 to 
#define E0_AUTO_FAN_PIN FAN2_PIN
and re-compiled it and now the fan plug work when the extrude hits the pre-determined temp.

torsoreaper, 2019-08-30:
Where are you finding the pin numbers?  I am trying to figure out if I can map the 12/24v in the top left to some kind of fan control in marlin but I don't know what pin that is or if it's simply a fixed voltage.

Thank you...

andracowolf, 2019-08-30:
I found the pin numbers here
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/hardware/SKR-V1.3-PIN.pdf

 The one next to the bed heaters is a the one that that I posted about. I
think the one next to the X axis is a always hot pin just full voltage.

torsoreaper, 2019-08-30:
Thanks man.  I had seen that one but I was hoping beyond all hope that I was missing something.  I have that plugged into a thin 80mm fan to cool the board and was hoping to map that to be a board cooling fan in marlin.

torsoreaper, 2019-08-31:
@andracowolf 

Just wanted to follow up in case anyone sees this thread.  I was able to link pin P2_04 (the 2nd hotend 12/24v) to my fan and enable #define USE_CONTROLLER_FAN in order to have my SKR board case fan only turn on when I'm printing.  Also from inside marlin you can set the max speed using #define CONTROLLERFAN_SPEED 255