bigtreetech/BIGTREETECH-SKR-V1.3 #74: Unable to access micro sd or through usb to octoprint

Silithas, 2019-08-31:
I use marlin 2.0.0 and here's my settings. I am unable to access the sd card nor over usb to octoprint. No errors, just nothing happens.

https://www.dropbox.com/s/f4gtcewkc7q05i2/Marlin.zip?dl=0

stoky996, 2019-09-01:
yea same here it refuses to show up on my computer when plugged in via usb micro sd card works fine though when i connect it via sd card reader

stoky996, 2019-09-01:
update tried installing smoothieware onto the board it shows up in windows but does not on my screen will try messing around with it.

stoky996, 2019-09-01:
fixed give this a shot grab smoothieware and install it onto the skr v1.3 try to connect it to your computer for now once smoothiware is installed using atom and platform io re flash a fresh copy of marlin onto the board and it should work made mine work.

here is a tutorial from teaching tech for marlin and manually install different firmwares.
https://www.youtube.com/watch?v=GNGN2iSQ5j4

Thanh-Viet, 2019-09-11:
I got mine with 

#define SDCARD_CONNECTION ONBOARD

in _adv.h

Silithas, 2019-09-11:
That didn't work for me. It was the first thing i tried.

XOIIO, 2020-02-03:
I have this issue too, can't believe they do such a shit job supporting their products.