bigtreetech/BIGTREETECH-SKR-V1.3 #75: Skr 1.3 TMC2208V3 Uart Issues

chopdi, 2019-09-03:
Hi .  I wired everything like shown up but get  TMC Connection Error in Display.  Checked via Pronter  3 TMC2008 get a Connection But  the 4. Has bad Response  All to low.  Can sw Help pls . I used BTT 2008 v3 Uart  and BTT skr1.3  compailed with atom.  TmC Lib up2date 

XY and EO Work...but Z not

Changed steppers but problem is still there.  Must be  a MB issue

Best reg.

tvixen, 2019-09-07:
So you changed the stepper TMC2208 from the X-axis to the Z-axis, and you still have the problem.
So the TMC2208 is ok.
Could you upload your configuration.h file

chopdi, 2019-09-07:
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020000

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
// config/examples/SCARA and customize for your machine.
//

[... 2121 lines cut ...]

/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// Delay (in milliseconds) before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Allow servo angle to be edited and saved to EEPROM
//#define EDITABLE_SERVO_ANGLES

chopdi, 2019-09-07:
Pronter:

M122 :

Printer is now online.
>>> m122
SENDING:M122
		X	Y	Z	E
Enabled		true	true	false	false
Set current	760	760	760	900
RMS current	734	734	1325	887
MAX current	1035	1035	1868	1251
Run current	23/31	23/31	23/31	28/31
Hold current	11/31	11/31	11/31	14/31
CS actual	11/31	11/31	0/31	14/31
PWM scale	39	40	0	16
vsense		1=.18	1=.18	0=.325	1=.18
stealthChop	true	true	false	true
msteps		16	16	256	16
tstep		max	max	0	max
pwm
threshold	98	98	10541	65
[mm/s]		100	100	3	30
OT prewarn	false	false	false	false
OT prewarn has
been triggered	false	false	false	false
off time	4	4	0	4
blank time	24	24	16	24
hysteresis
-end		2	2	-3	2
-start		1	1	1	1
Stallguard thrs
DRVSTATUS	X	Y	Z	E
stst		*	*		*
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:0B:00:00
		Y	0xC0:0B:00:00
		Z	0x00:00:00:00	 Bad response!
		E	0xC0:0E:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... Error: All LOW
Testing E connection... OK

chopdi, 2019-09-07:
H nd ADV

chopdi, 2019-09-10:
No one?

tvixen, 2019-09-10:
#define X_DRIVER_TYPE TMC2208  OK
#define Y_DRIVER_TYPE TMC2208  OK
#define Z_DRIVER_TYPE TMC2208  OK
//#define X2_DRIVER_TYPE A4988 
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2208 OK
#define E1_DRIVER_TYPE DRV8825   NOT OK.  Comment out
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988

this should work for your config.h

tvixen, 2019-09-10:
your CONFIGURATION_ADV.H is also wrong. Please paste it.

tvixen, 2019-09-10:
Also please write the type of Ender 3 you are using (Pro, X or normal) This has to do with the voltage settings.

chopdi, 2019-09-11:
Thank you.  I will check after work.

Just the normal one. 
Vrefs r 760 for xyz and 900 for e

chopdi, 2019-09-11:
#define X_DRIVER_TYPE TMC2208 OK
#define Y_DRIVER_TYPE TMC2208 OK
#define Z_DRIVER_TYPE TMC2208 OK
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2208 OK
#define E1_DRIVER_TYPE DRV8825 NOT OK. Please Comment out
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988

this should work for your config.h

So u agree everything is okay and it should work?

E1 is commentet out

tvixen, 2019-09-11:
Yes it should work.  

Also please write the type of Ender 3 you are using (Pro, X or normal) This has to do with the voltage settings to the TMC's  they run better with 24V as you have written in AdV.config.h

But just to be shure, measure your supply from the PS to the board. It should be 24volt.
If not, please correct value in advance.config.h

chopdi, 2019-09-11:
Just a normal ender 3.

chopdi, 2019-09-11:
I will check voltage later after work but should be 24

chopdi, 2019-09-11:
24,2 V

chopdi, 2019-09-12:
It is not working  .

tvixen, 2019-09-12:
Take out all your drivers and check them one by one.
Check your dip's on the MB, then
#define X_DRIVER_TYPE TMC2208
//#define Y_DRIVER_TYPE TMC2208
//#define Z_DRIVER_TYPE TMC2208
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
//#define E0_DRIVER_TYPE TMC2208
//#define E1_DRIVER_TYPE TMC2208
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988

1. Use the same TMC2208 in all the axes
2 put it in X, test code #define X_DRIVER_TYPE TMC2208 //#define Y_DRIVER_TYPE TMC2208
3 put it in Y, test code //#define X_DRIVER_TYPE TMC2208 #define Y_DRIVER_TYPE TMC2208
4 put it in Z, test code //#define X_DRIVER_TYPE TMC2208 //#define Y_DRIVER_TYPE TMC2208 #define Y_DRIVER_TYPE TMC2208

Run the test command every time: M122

If still no contact to Z axis, you must find you multimeter and check the connection from the cpu to Z socket. Its has been seen before, that a connection is broken inside the PCB.

chopdi, 2019-10-01:
Mb is broken.  I buy a new one and it works ofb. 

Payed it twice :/

fiferboy, 2019-10-01:
It looks like the same problem is happening to me with the exact same symptoms except the Y axis is throwing the fault. Everything else is identical to this case.

Things I have confirmed:

- board is powered by 24V and happily working other than the one UART
- 2208 V3 is properly modified from the factory with a 0-ohm patch over J2
- switching another 2208 to the slot does not solve the problem

Steps I have tried:

- switching another 2208 to the Y axis
- shorting various pins under the stepper (found in other threads)
- trying firmware directly from the manufacturer
- compiling Marlin with different versions of the TMCStepper library

I am coming to the conclusion that there could be a fault in my board as well - thank you for confirming that was the root cause @chopdi

tvixen, 2019-10-01:
> Mb is broken. I buy a new one and it works ofb.
> 
> Payed it twice :/

Im sorry to hear that, but at least you found the fault.  
BIGTREETECH will refund you, as they have done it for other users.

thomasdr1, 2019-12-07:
I have the Bigtreetech TMC2208 V3 which should work in UART mode.
The jumpers and Marlin 2 are set according to instructions.
But not a single TMC2208 works in UART mode.

ZhuDaHai, 2020-02-28:
> Yes it should work.
>
> Also please write the type of Ender 3 you are using (Pro, X or normal) This has to do with the voltage settings to the TMC's they run better with 24V as you have written in [...]

Where is 'advance.config.h' / 'AdV.config.h' ?
I can't find that file anywhere
Thanks

KixMan28, 2020-04-14:
Today I upgraded the main board to BTT SKR 1.3 with BTT TMC2208 V3.0 in UART mode, with the latest Marlin 2.0.X firmware and MKS TFT28 V4.0 display. I have multiple problems with this setup:
- if I run M122 in Pronterface I get ALL LOW error message on all axis
- if I move the motors from Pronterface they move 20 mm instead of 10 mm
- if I auto home X first the Z goes up 20 mm and the it homes X.
- the TFT is connected on AUX1 on the SKR1.3 but when I switch it on there is nothing on the display.
- if I power up the display with 12V I can move the axis (with the errors mentioned above) and heat up the nozzle and bed correctly.
- I have set the serial baud rate to 152000 or 250000 on both boards, but no result.
- I have watched several YT videos on how to install SKR1.3-TMC2208 (including the ones from here) and I have googled the problems but I could not find solution for any of them.
- when I powered on the SKR1.3 for the first time it beeped twice. Now is not beeping anymore. When powered on the red led is on.
What I have done already:
On the board I have set the jumper on INT-5V. I have jumpers on the corresponding UART pins (X, X, Z, E0). All the other jumpers are removed. I didn't soldered the pads on the bottom of the TMC2208. In Marlin the drivers are set to TMC2208 and the e-steps are 80, 80, 400, 93

Defaultsound, 2020-04-15:
> Today I upgraded the main board to BTT SKR 1.3 with BTT TMC2208 V3.0 in UART mode, with the latest Marlin 2.0.X firmware and MKS TFT28 V4.0 display. I have multiple problems with this setup:
>
> * if I [...]

I am experiencing the exact same problem, only I have the SKR 1.4 Turbo. Jumpers are set correctly, but also seeing motors run twice as fast or double the distance. M122 reports all are low. Seems like drivers could be at fault here?

KixMan28, 2020-04-15:
It seems that mine too are simple ones and not UART, at least based on my order details. I will check them visually tomorrow.

Defaultsound, 2020-04-16:
> It seems that mine too are simple ones and not UART, at least based on my order details. I will check them visually tomorrow.

Yup so soldered my J2 pads today and sure enough now M122 reports back correctly. Steps are also correct. https://github.com/bigtreetech/BIGTREETECH-TMC2208-V3.0/issues/1 also this issue has a clearer image to show you what to solder. I middle to bottom pad worked for me, or the image on the left on that issue.

KixMan28, 2020-04-16:
@Defaultsound So you soldered these pads (see red area in the image)?

[image]

Defaultsound, 2020-04-16:
@KixMan28  Yes I soldered those two

KixMan28, 2020-04-16:
@Defaultsound OK, thank you. I will solder them tomorrow.

KixMan28, 2020-04-17:
I have soldered the pads on the bottom of the TMC2208 drivers and after several plugging in and out, finally all drivers were recognized by my SKR1.3 board.

SkyAce-89, 2021-02-24:
Hi,
I have simmilar issue and had to use "M122 I" multiple times (once per each uart driver used) with short delay between calls.

My setup/configuration:
SKR 1.4 Turbo
X,Y,Z axes with mix of TMC2208 (watterott) and TMC2209 (BTT v3.0)
RasPi+Octoprint with PSU control.

SKR 1.4 MCU is powered from USB.
Octoprint switches the 24V PSU ON at the start of the print.

Route cause:
With TMC drivers powered from 24V and MCU powered from USB 5V, 24V power cycle breaks the UART communicatio with MCU.
- Allow TMC_DEBUG in Marlin and run "M122". See the driver registration lines. 

Trial 1:
Usb powered MCU. 24V PSU is OFF.
M122 will display "bad response" from UART drivers.

Trial2:
USB powered MCU. 24V PSU is ON.
M122 will display active UART drivers, but driver registration has same values for any two or more drivers. Those drivers will run at standalone mode or will not be running at all.
execute "M122 i" command: driver registration will be reset for the first non-functioning driver.

i used following chunk of gcode at the start of my print file right after m502:
```
M122 i
G6 s2000
```