bigtreetech/BIGTREETECH-SKR-V1.3 #79: 3D Touch (Geeetech) not working with Marlin

john4242, 2019-09-17:
I have an SKR 1.3 board and a 3DTouch (Geeetech "original") sensor. I have the same type of sensor working with Marlin on a GT2560 A+ board, although not the same piece.

What works:

    The servo part and the 3D Touch control is working well: I can stow and unstow with M commands, pins move, it reacts to sensing (pulls the pin back). The wiring for the servo is easy: there is a servo connector on the SKR 1.3 board, I just had to make sure the pin order was correct (+5V, GND, Servo).

    However, I have just struggled with the actual sensing. It's not getting triggered. I compiled Marlin with the PIN debug option, and there is actually one constellation which "almost" works:
    -- I plugged the touch sensors into Z MIN (black to GND, middle pin, white to signal pin)
    -- Z_MIN_ENDSTOP_INVERTING = false // as default
    -- ENDSTOPPULLUP_ZMIN_PROBE defined
    -- ENDSTOPPULLUP_ZMIN defined

I have tried inverting as true and also as false (the true is tricky because BLTOUCH define overrides it in another file, but I removed that override).

I got as far as:

    If inverting = true, then M43 S shows:

Servo probe test
. using index:  0, deploy angle: 10, stow angle:   90
. Probe Z_MIN_PIN: 57
. Z_MIN_ENDSTOP_INVERTING: true
. Check for BLTOUCH
. Deploy & stow 4 times
WARNING: INVERTING setting probably backwards.
** Please trigger probe within 30 sec **
. Pulse width (+/- 4ms): 6
FAIL: Noise detected - please re-run test
ok

If inverting = false, then M43 S returns:

Servo probe test
. using index:  0, deploy angle: 10, stow angle:   90
. Probe Z_MIN_PIN: 57
. Z_MIN_ENDSTOP_INVERTING: false
. Check for BLTOUCH
= BLTouch Classic 1.2, 1.3, Smart 1.0, 2.0, 2.2, 3.0, 3.1 detected.
** Please trigger probe within 30 sec **
. Pulse width (+/- 4ms): 4
FAIL: Noise detected - please re-run test
ok

The interesting bit as that it is not noise what it's detecting, that's my actual touch detected.
I'm starting to be out of ideas, spent hours on the net googling, got some tips, none worked.

I also tried to move it to the ZMAX pin, and repeat the tests.
It is close to working, but with that "noise detected" error.

Servo probe test
. using index:  0, deploy angle: 10, stow angle:   90
. Probe Z_MIN_PROBE_PIN: 1336
. Z_MIN_PROBE_ENDSTOP_INVERTING: true
. Check for BLTOUCH
. Deploy & stow 4 times
WARNING: INVERTING setting probably backwards.
** Please trigger probe within 30 sec **
. Pulse width (+/- 4ms): 6
FAIL: Noise detected - please re-run test
ok
Reporting endstop status
x_min: open
y_min: open
z_min: TRIGGERED
z_probe: open
ok
Servo probe test
. using index:  0, deploy angle: 10, stow angle:   90
. Probe Z_MIN_PROBE_PIN: 1336
. Z_MIN_PROBE_ENDSTOP_INVERTING: true
. Check for BLTOUCH
. Deploy & stow 4 times
WARNING: INVERTING setting probably backwards.
** Please trigger probe within 30 sec **
. Pulse width (+/- 4ms): 4
FAIL: Noise detected - please re-run test
ok

I have two 3D Touch sensors (Geeetech), both exhibit the same effect. Any ideas?

john4242, 2019-09-19:
Okay, I have put together some Arduino circuit to test the sensor.
It confirms that the output pulse (HIGH level) only happens for 4-6 milliseconds, and not 15 milliseconds or more as Marlin expects.
If anyone wants to test and play with Arduino, I have attached the arduino code, could be useful for testing other sensors, too.
[main.cpp.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/3633416/main.cpp.txt)

lococnc, 2019-09-30:
So how do we change the 15ms Marlin expectation? Is this in the code somewhere where we can shorten this?

john4242, 2019-09-30:
I thought you need to change it. But actually it's just a totally misleading message.
Marlin will work fine even if you have a (better) clone with a 5 ms long pulse.
So you can ignore that error message. I am thinking of submitting a patch to marlin so it doesn't mislead people.