bigtreetech/BIGTREETECH-SKR-V1.3 #80: LCD2004 works bad

Decygnus, 2019-09-17:
I have LCD 2004. While printer is printing when I push on the encoder and screen is hanging without any information! Print is continuing, but screen doesn't answer for push ups or rotating. After some time it shows the info screen. All parameters are updating in real time, but when i push on encoder....the same hanging. What's wrong?

This problem appears only when printing

greezmen, 2019-09-17:
set block buffer size  to 64

Decygnus, 2019-09-17:
Thank you for answer.
Is it right in Configuration_adv:

//===========================================================================
//================================= Buffers =================================
//===========================================================================

// @section hidden

// The number of linear motions that can be in the plan at any give time.
// THE BLOCK_BUFFER_SIZE NEEDS TO BE A POWER OF 2 (e.g. 8, 16, 32) because shifts and ors are used to do the ring-buffering.
#if ENABLED(SDSUPPORT)
  #define BLOCK_BUFFER_SIZE 64 // SD,LCD,Buttons take more memory, block buffer needs to be smaller
#else
  #define BLOCK_BUFFER_SIZE 64 // maximize block buffer
#endif

both of them?  Doesn't it make any bad influence on other parameters or print quality ?

tvixen, 2019-10-07:
> both of them? Doesn't it make any bad influence on other parameters or print quality ?

It will not have any influence on the print quality, but the response of the printer will be longer, as the buffer has to be empty, before it reacts. This is definitely a problem for Octoprint, when you try to cancel a print.