bigtreetech/BIGTREETECH-SKR-V1.3 #83: firmware.bin not converting to FIRMWARE.cur

kurtisane, 2019-09-22:
Hi guys, 
I'm having issues for the last two days. 
I'm compiling Marlin and putting the firmware.bin on the SD Card. Before it was renameing the fiel to FIRMWARE.cur and everything was fine. 

I did not change anything on the Marlin Config and it is not renameing anymore. I have the feeling the bootloader might be corrupted or so. I cleaned the sd card slot with some 99% alcohol. I dont see any dirt in there. Swaped to a different SD Card also. They are both FAT formated. I also tried FAT32.

I have a BLTouch connected to the Board and the Pin is getting pushed out and pulled back. This behavior was also present before this happend. 

Can someone tell me how to flash the bootloader or any idea what might be the issue ? 

Regards !

K34nu, 2019-10-03:
Would also like to say that i'm having exactly the same issue. Seems like a corrupted bootloader, though i'm not able to find a copy of it to re-flash.

tvixen, 2019-10-07:
Remember to reset the SKR 1.3 board twice on the reset button.

karabas2011, 2019-10-31:
Why twice?

tvixen, 2019-10-31:
Because 1.st time the firmware will be loaded into the CPU, and 2.nd time the CPU will start up with the new firmware installed. Don't know why the bootloader isn't doing it in one step, but properly a code tester not being awake before they released the chip ;O)

karabas2011, 2019-10-31:
I use sound command M300 in START_COMMANDS
so I hear end of boot even if screen blank or something

hexcorsist, 2019-11-25:
I'm having the exact same problem tried 2 sd cards and firmware.bin is not being renamed, does anybody have a solution?

Chron-O-John, 2019-12-15:
I also had the same problem.
I tried anothe, empty(!!) SD-Card and it worked. It was 16GB big and FAT32 formatted

bmangodt, 2019-12-29:
Same issue here... was working fine in the beginning, wanted to update to the official release of Marlin 2.0... Inserted the sd card into the computer and it was not read. The sd card that comes with the board has the tendency of going corrupt, so I tried a new one... no luck. Tried 3 different cards of different brands (Kingston, Sandisk and Samsung) with different sizes (2GB, 16GB and 64GB) all formatted in FAT32. Tried both empty and preloaded with a dummy FIRMWARE.CUR (yes I checked the extension) and either with or without a firmware.bin file present on the card. No luck!!

tvixen, 2019-12-30:
Try keep the reset pressed down for 3 sec
If this doesn't work, try copy a .stl file to the card, and then print it from the card, just to see if the card is corrupt or not. It the printer can print from the card, it's not the card failing.

bmangodt, 2019-12-30:
Managed to fix it yesterday... No idea what I did differently but I formatted a 16gig SD card to FAT32. Connected to the printer via usb (skr to 5v usb power with the jumper pins) and uploaded firmware.bin through usb. Pulled usb out, plugged it back in and it worked. So my suggestion: keep on trying

BIZUx, 2020-05-31:
Having similar issues with 2.0.x. 
Even if I try connecting through a USB to my pc, still nothing (I've tried with 3 different cords).

txt4nk, 2020-06-01:
These boards are extremely picky with the SD cards. I have come to adapt a format on every new firmware flash (started this maybe 6 months ago?) and I have not had this issue come up since. It must be FAT32 (not FAT, every time I ever had this issue, when inserting the card and going to the format menu, it was always FAT instead of FAT32). Went digging through my pile if SD's and found one with this issue, here's a snip of it below, along with the format command and how it should be configured. 

Once you do this, this issue will be a thing of the past.
[image]

tvixen, 2020-06-01:
Do your self a favor, and buy a new quality 16Gb SD card. Instead of these 128Mb cheeeeep ones!
Then you will have no more problems with the SD card.

txt4nk, 2020-06-01:
That too, I run 8GB samsung class 10 cards exclusively now. Still in the habit of formatting before every firmware flash though.

BIZUx, 2020-06-02:
Thanks for your replies!
However, I'm thinking the problem lies beyond the SD card, as I've already tried formatting a good quality 32 GB microSDHC to FAT32 through diskpart. 
Do you think maybe it could be a faulty bootloader?

Cimmarron, 2021-04-17:
Hi. It Happens to me.
 I solved using a Micro SD Card with less capacity. Maybe there is a max GB Micro Sd reading capacity in the SKR V1.3

Person-maker, 2021-06-28:
Hi. I had the same problem and I have solved it really easily. I had just formatted the card of 128Mb, I think, from EXFAT to FAT32 and I had reset the skr 1.4 twice and now it works, and with the same card of the BTT.

MIR3D-Leeds, 2024-04-17:
Yes, I know this is an ancient issue, but it may help someone else who lands here. Use SD Card Formatter utility. https://www.sdcard.org/downloads/formatter/sd-memory-card-formatter-for-windows-download/ 

Format SD Card, copy firmware.bin to it, and continue as normal.