bigtreetech/BIGTREETECH-SKR-V1.3 #84: Not reaching 5Volts.

REDxFROG, 2019-09-22:
I have connected a BLTouch AND a 3DTouch to the printer. They kind of work. But not really.

Measured the 5V pins for the SERVO and END STOPS. They are not 5V. Only ~4.88V.
Is that normal? How to increase the voltage?

I remember some time ago, I plugged something in the wrong way and the board would beep, the display would only show backlight. Is it broken?
The printer/board runs fine, but every time I do a cold start on the BLTouch/3DTouch, it will not retract/do the self test. Only when I move the sensor with my finger once, and turn it on again, it works. But not reliable enough..

dant220, 2019-09-22:
TTL logic levels are listed here:
https://learn.sparkfun.com/tutorials/logic-levels/all

4.8V is within spec.

tvixen, 2019-10-07:
Power supply is ok. Try a 22uf over the 5v supply or a new BLTouch.