bigtreetech/BIGTREETECH-SKR-V1.3 #85: Can't copile marlin to the board.

LMADoneur, 2019-09-25:
> Executing task: C:\Users\doneu\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; framework: arduino; board: nxp_lpc1768)
-----------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.0.2 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.80KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1), framework-arduino-lpc176x 0.1.3
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph 
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4
|-- <TMCStepper> 0.5.0
|-- <Adafruit NeoPixel> 1.2.4
|-- <SailfishLCD>

Unable to find destination disk (a bytes-like object is required, not 'str')
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy 
the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_sw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\watchdog.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/MarlinConfigPre.h:36,
                 from Marlin\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_sw_spi.cpp:58:
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration.h:491:33: warning: multi-line comment [-Wcomment]
 #define PID_FUNCTIONAL_RANGE 10 // If the temperature difference between the target temperature and the actual temperature \
                                 ^
Compiling .pio\build\LPC1768\src\src\HAL\shared\HAL_spi_L6470.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/MarlinConfigPre.h:42,
                 from Marlin\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_sw_spi.cpp:58:
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:1017:34: warning: multi-line comment [-Wcomment]
 #define SDSORT_CACHE_VFATS 2     // Maximum number of 13-byte VFAT entries to use for sorting. \
                                  ^
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:1284:39: warning: multi-line comment [-Wcomment]
 #define DOUBLECLICK_MAX_INTERVAL 1250 // Maximum interval between clicks, in milliseconds. \
                                       ^
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:1599:46: warning: multi-line comment [-Wcomment]
 #define PAUSE_PARK_RETRACT_LENGTH 2          // (mm) Initial retract. \
                                              ^
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:1603:46: warning: multi-line comment [-Wcomment]
 #define FILAMENT_CHANGE_UNLOAD_LENGTH 100    // (mm) The length of filament for a complete unload.    \
                                              ^
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:1608:46: warning: multi-line comment [-Wcomment]
 #define FILAMENT_CHANGE_SLOW_LOAD_LENGTH 0   // (mm) Slow length, to allow time to insert material. \
                                              ^
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:1612:46: warning: multi-line comment [-Wcomment]
 #define FILAMENT_CHANGE_FAST_LOAD_LENGTH 0   // (mm) Load length of filament, from extruder gear to nozzle. \
                                              ^
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:1617:41: warning: multi-line comment [-Wcomment]
 #define ADVANCED_PAUSE_PURGE_LENGTH 50  // (mm) Length to extrude after loading.                               \
                                         ^
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:2499:29: warning: multi-line comment [-Wcomment]
 #define I2CPE_ENCODER_CNT 1 // The number of encoders installed; max of 5 \
                             ^
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:2504:48: warning: multi-line comment [-Wcomment]
 #define I2CPE_ENC_1_TYPE I2CPE_ENC_TYPE_LINEAR // Type of encoder:  I2CPE_ENC_TYPE_LINEAR -or- \
                                                ^
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:2506:48: warning: multi-line comment [-Wcomment]
 #define I2CPE_ENC_1_TICKS_UNIT 2048            // 1024 for magnetic strips with 2mm poles; 2048 for  \
                                                ^
[... 153 lines cut ...]
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:2547:34: warning: multi-line comment [-Wcomment]
 #define I2CPE_TIME_TRUSTED 10000 //Compiling .pio\build\LPC1768\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
 After an Compiling .pio\build\LPC1768\src\src\HAL\shared\eeprom_i2c.cpp.o
encoder fault, there must be no further fault \
                                  ^
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:2513:51: warning: multi-line comment [-Wcomment]
 #define I2CPE_ENC_1_EC_THRESH 0.10                // Threshold size for error (in mm) above which the     \
                                                   ^
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:2578:32: warning: multi-line comment [-Wcomment]
 #define MAX7219_ROTATE 0       // Rotate the display clockwise (in multiples of +/- 90Â°) \
                                ^
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:2547:34: warning: multi-line comment [-Wcomment]
 #define I2CPE_TIME_TRUSTED 10000 // After an encoder fault, there must be no further fault \
                                  ^
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:2591:39: warning: multi-line comment [-Wcomment]
 #define MAX7219_DEBUG_PLANNER_QUEUE 0 // Show the current planner queue depth on this and the next LED matrix row \
                                       ^
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:2578:32: warning: multi-line comment [-Wcomment]
 #define MAX7219_ROTATE 0       // Rotate the display clockwise (in multiples of +/- 90Â°) \
                                ^
Marlin\src\HAL\HAL_LPC1768\u8g\../../../inc/../../Configuration_adv.h:2591:39: warning: multi-line comment [-Wcomment]
 #define MAX7219_DEBUG_PLANNER_QUEUE 0 // Show the current planner queue depth on this and the next LED matrix row \
                                       ^
In file included from Marlin\src\HAL\shared\../../inc/MarlinConfigPre.h:42,
                 from Marlin\sr

tvixen, 2019-09-30:
You forgot to turn on the SKR 1.3 or connect it to the PC.

CsepregiArtur, 2019-10-03:
comply it first and after that move it to the sd card, sart your printer and ready